synthetos/TinyG #117: Planner queue size for arc chords

dougle, 2015-01-02:
I know this is going to be met with a fair bit of resistance but trying to track the progress of an ngc file that contains G02 and G03 commands is near impossible, even with the triple queue reports from 0.97.

I have been running G2 X100 Y100 I150 J150 alone and this fills the queue buffer.

I understand that arcs are split into chords, but from a developers point of view having qo in the status  count each chord is no use, if the queue reports could collate all of the chords down into one command in terms of queue reports that would help properly track progress.

aldenhart, 2015-01-18:
It's for this reason that queue reports are not as useful to track progress as we had hoped. I think the best way to do this now is to manage the serial buffer size using the buffer remaining command (rx) and the "bytes-removes" value in the JSON footer. There are a few other alternatives, but this is the one most people are using at this point.

dougle, 2015-01-20:
Just so i'm clear, the process you're suggesting is:

Total up the number of bytes to be sent to the serial port.
Keep a running total of the footer element "RX_received".
Send commands normally with either XonXoff or RTS/RTS.
Divide one by the other and multiply by 100 for the percentage.

aldenhart, 2015-01-21:
I believe so. In short, there are 254 bytes available in the serial RX queue when it is empty. The {rx:n} or $rx command will tell you the exact number. You can put in up to that amount before you get an overflow. As the interpreter pulls commands out of the buffer it reports back how many bytes it consumed in the footer. You can use this as a running total to tell how much more the buffer can accept. You can also issue periodic RX commands to get the absolute space available in the buffer.

closed 2015-01-28 by aldenhart