synthetos/TinyG #118: [HELP] Remapping Serial0 to Serial1 on ATXMEGA192

Ultrone, 2015-01-04:
I broke the serialport on my atxmega192 (i put 5V to input RX…).

How i do for change serial port configuration and compile Xboot and TinY projects, on AVRstudio 4?

I would change the SerialPort0 (pinPortC 2,3) on C, with SerialPort1 (pinPortC 6, 7) still on C

(i have AVRS 4.18 + SP3)

Please explain step by step.

Thanks!

ril3y, 2015-01-04:
Ultrone,

This is well outside an "issue"  I suggest you look at the developer pages.
https://github.com/synthetos/TinyG/wiki/Introduction-to-the-TinyG-Code-Base
Also there is no way to tell if you blew anything else in the internals of the uC.

closed 2015-01-04 by ril3y

Ultrone, 2015-01-05:
This is over my skill. I should have to study the whole project.

Give me a hint faster?

There is some configuration file (.H) where you can simply change the number of the serial port?

Thanks