synthetos/TinyG #119: Configurable spindle pin reversable

Cinezaster, 2015-01-06:
I'm converting an old Roland CAMM-3 machine.
M5 starts the spindle.
M3 or M4 stops the spindle.
Would be nice if we can revert the spindle pin output with a $ command.

I have to dial in the speed on a potentiometer. And no options for the direction.

mikest, 2015-01-06:
that's backwards, and dangerous.

put an inverter in front of your active low spindle enable. then, when you get a tinyg power failure your spindle won't come on...

Cinezaster, 2015-01-07:
Indeed, it sounds like a bad idea and I didn't put thought to it.

closed 2015-01-07 by Cinezaster

ejluttmann, 2015-01-07:
I modified the firmware to allow setting a $sal command to switch to active low on the spindle pin (see https://github.com/ejluttmann/TinyG/commit/6469260c2338d080c923fd623808e542a38e7189) in order to be compatible with the SuperPID 2.  When I tried different power on/off configurations, I didn't see any problems ... but that might be because of the SuperPID 2 board.

Cinezaster, 2015-01-07:
I might try that. Right now it is indeed dangerous because the spindle starts spinning on start-up of the board. A bit scary if I'm honnest.

ejluttmann, 2015-01-07:
That was my issue as well during startup - I also wired the SuperPid 2 so that I can switch between a manual on/off using potentiometer vs using the tinyg board, just in case.  But it's nice to have the tinyg turn on and off the spindle.

Cinezaster, 2015-01-07:
@ejluttmann I have no bootloader installed on my brand new TinyG, so have to fix me a spi programmer before I can update my firmware.

ril3y, 2015-01-08:
If you got your TinyG from us it has a bootloader.  Unless it was purchased
2 years ago :)

troisiemetype, 2015-01-08:
I have the same problem with mine. I've ordered a two relay module (the kind you can find everywhere you can find arduino) that drives the two plugs for spindle and cooling, but this module enable the relays on low state, where the TinyG enable on high state (spin=1 for M3, spin =0 for M5).
It would be easy to create a NOT gate before each pin of the module with two op-amp.
Another solution would be, as the guy I bought the module suggest me, to add on each pin of the module a pullup-resistance to drive it HIGH, and connect the spindle on the NC relay output, instead of the NO.

Cinezaster, 2015-01-08:
@ril3y I just purchased this in december. No spindle dir led is flashing
for 3seconds after I push reset button. Also $boot=1 will reset the device
but it starts immediately. I tried to write the firmware but get always an
error message stating that the device is not ready. Also I can write
immediately to the serial bus and get a respons from the program after a
$boot=1. That shouldn't happen if it is bootloading mode.

-- 
AppSaloon - Webdesign & Appdesign op maat
Toon Nelissen
T: +32 485 832 970
E: toon@appsaloon.be
W: www.appsaloon.be
Bampslaan 21 3.01, 3500 Hasselt

De informatie in dit bericht is uitsluitend bedoeld voor de geadresseerde
en kan vertrouwelijke en/of bevoorrechte gegevens en/of door
intellectuele-eigendomsrechten beschermde informatie bevatten.  Als u niet
de geadresseerde bent, gelieve dit bericht te verwijderen en de afzender te
verwittigen. U mag dit bericht niet gebruiken, wijzigen, kopiëren of
verspreiden, noch de inhoud ervan bekendmaken aan een derde.
De veiligheid of juistheid van e-mailberichten kan niet gegarandeerd
worden, vermits de informatie onderschept, verbasterd of vernietigd kan
worden, zoek kan raken, te laat of onvolledig kan aankomen of virussen kan
bevatten. APPSALOON aanvaardt geen enkele aansprakelijkheid voor verlies of
schade die op enigerlei wijze te wijten is aan het gebruik van het medium.
Eventuele standpunten of meningen in dit bericht zijn die van de auteur en
geven niet noodzakelijk die van APPSALOON weer. Bijgevolg bindt dit
e-mailbericht niet, tenzij het een uitdrukkelijke andersluidende verklaring
van een gemachtigde vertegenwoordiger bevat.

lagnat, 2020-08-24:
Old thread but I'm having the same issue and I'm not verse enough with digital electronics to know how to proceed.  I've got a relay board that has a NO and NC pin.  It seems that the NO pins connect when the trigger is pulled low, which means the spindle starts to spin shortly after TinyG bootup.  If I use the NC pins, then the spindle start up right away and then stops when the TinyG boots up.  This is currently my best option.

It was suggested that by allowing the TinyG to support enabling the spindle with a high level signal, that a TinyG power failure would create a dangerous situation.  In my case, it's the opposite.  I have a dangerous situation by not allowing me to reverse the signal.  Either I get a spinning spindle for a few seconds during power up, or I get a spindle that would start up if the TinyG lost power.

IMO this would be a useful feature despite enabling scenarios where people could be foolish.