synthetos/TinyG #120: TinyG G18 XZ Plane Seems to be Reversed

thrushd, 2015-01-08:
Hello,

I recently upgraded my custom CNC to TinyG. So far I am absolutely loving the intuitive settings and ease of use, unfortunately a small issue popped up. When my machine goes to do an arc (using G2 and G3, but it also happens in radius mode) in the G18 plane it will move in the opposite direction it is intended (CW instead of CCW). It only seems to happen in the G18 plane. I noticed this when trying a simple facing operation for my vacuum bed, the specific G18 arcs are being used for the lead ins. They were very small though so it was hard to tell at first. This prompted me to make a quick test piece to experiment on, just a 50mm x 50mm block. 

[image]

Did the same basic facing operation but this time made the lead in and out radii much bigger, 10mm. It is very noticeable now. Also loaded it into the HSMWorks backplotter and played with the code. Reversing the G2 and G3 shows what my machine is actually doing:

[image]

But when I use that modified reversed code on the machine it behaves great. So it seems to me that the G18 plane is reversed on the TinyG. I have played with all the settings I can think of, nothing changes. Used a variety of post processors and got the same results so I have been sticking to the built in "generic tinyg". Completely out of ideas and haven't been able to find anyone else with this issue. Any help or insight greatly appreciated!

**Details:**

CAD: Solidworks 2014 Education Edition
CAM: HSMWorks 2014

TinyG Settings Dump:

```
[fb]  firmware build            438.02
[fv]  firmware version            0.97
[hp]  hardware platform           1.00
[hv]  hardware version            8.00
[id]  TinyG ID                    3X3566-SRM
[ja]  junction acceleration  100000 mm
[ct]  chordal tolerance           0.0100 mm
[sl]  soft limit enable           0
[st]  switch type                 0 [0=NO,1=NC]
[mt]  motor idle timeout         30.00 Sec
[ej]  enable json mode            0 [0=text,1=JSON]
[jv]  json verbosity              1 [0=silent,1=footer,2=messages,3=configs,4=linenum,5=verbose]
[js]  json serialize style        1 [0=relaxed,1=strict]
[tv]  text verbosity              1 [0=silent,1=verbose]
[qv]  queue report verbosity      2 [0=off,1=single,2=triple]
[sv]  status report verbosity     1 [0=off,1=filtered,2=verbose]
[si]  status interval           250 ms
[ec]  expand LF to CRLF on TX     0 [0=off,1=on]
[ee]  enable echo                 0 [0=off,1=on]
[ex]  enable flow control         1 [0=off,1=XON/XOFF, 2=RTS/CTS]
[baud] USB baud rate              5 [1=9600,2=19200,3=38400,4=57600,5=115200,6=230400]
[net] network mode                0 [0=master]
[gpl] default gcode plane         0 [0=G17,1=G18,2=G19]
[gun] default gcode units mode    1 [0=G20,1=G21]
[gco] default gcode coord system  1 [1-6 (G54-G59)]
[gpa] default gcode path control  2 [0=G61,1=G61.1,2=G64]
[gdi] default gcode distance mode 0 [0=G90,1=G91]
[1ma] m1 map to axis              0 [0=X,1=Y,2=Z...]
[1sa] m1 step angle               1.800 deg
[1tr] m1 travel per revolution    5.0000 mm
[1mi] m1 microsteps               8 [1,2,4,8]
[1po] m1 polarity                 0 [0=normal,1=reverse]
[1pm] m1 power management         2 [0=disabled,1=always on,2=in cycle,3=when moving]
[2ma] m2 map to axis              1 [0=X,1=Y,2=Z...]
[2sa] m2 step angle               1.800 deg
[2tr] m2 travel per revolution    5.0000 mm
[2mi] m2 microsteps               8 [1,2,4,8]
[2po] m2 polarity                 0 [0=normal,1=reverse]
[2pm] m2 power management         2 [0=disabled,1=always on,2=in cycle,3=when moving]
[3ma] m3 map to axis              2 [0=X,1=Y,2=Z...]
[3sa] m3 step angle               1.800 deg
[3tr] m3 travel per revolution    5.0000 mm
[3mi] m3 microsteps               8 [1,2,4,8]
[3po] m3 polarity                 0 [0=normal,1=reverse]
[3pm] m3 power management         2 [0=disabled,1=always on,2=in cycle,3=when moving]
[4ma] m4 map to axis              3 [0=X,1=Y,2=Z...]
[4sa] m4 step angle               1.800 deg
[4tr] m4 travel per revolution  360.0000 mm
[4mi] m4 microsteps               8 [1,2,4,8]
[4po] m4 polarity                 0 [0=normal,1=reverse]
[... 117 lines cut ...]
[g59a] g59 a offset               0.000 deg
[g59b] g59 b offset               0.000 deg
[g59c] g59 c offset               0.000 deg
[g92x] g92 x offset               0.000 mm
[g92y] g92 y offset               0.000 mm
[g92z] g92 z offset               0.000 mm
[g92a] g92 a offset               0.000 deg
[g92b] g92 b offset               0.000 deg
[g92c] g92 c offset               0.000 deg
[g28x] g28 x position             0.000 mm
[g28y] g28 y position             0.000 mm
[g28z] g28 z position             0.000 mm
[g28a] g28 a position             0.000 deg
[g28b] g28 b position             0.000 deg
[g28c] g28 c position             0.000 deg
[g30x] g30 x position             0.000 mm
[g30y] g30 y position             0.000 mm
[g30z] g30 z position             0.000 mm
[g30a] g30 a position             0.000 deg
[g30b] g30 b position             0.000 deg
[g30c] g30 c position             0.000 deg
tinyg [mm] ok> 
```

Thanks!

aldenhart, 2015-01-18:
That's for the heads up and the detailed. This is a bug that should be corrected.

thrushd, 2015-01-20:
So this is just a bug in the current firmware? Nothing more I can go until the next release? Thanks!

aldenhart, 2015-01-21:
We'll try to get to this quickly and post an update for an experimental release if you are interested.

thrushd, 2015-01-21:
I would definitely be interested, just keep me posted, thanks!

aldenhart, 2015-02-09:
Hi. Checking back in after some work on the arcs - It looks like the G18 arcs are actually working correctly, unless I'm misinterpreting things. TinyG follows the NIST and Linux CNC definition of Gcode. From here:

http://linuxcnc.org/docs/html/gcode/gcode.html#sec:G2-G3-Arc 
"A circular or helical arc is specified using either G2 (clockwise arc) or G3 (counterclockwise arc) at the current feed rate. The direction (CW, CCW) is as viewed from the positive end of the axis about which the circular motion occurs."

So G18 G2... should trace a circle in the clockwise direction about Y as viewed from the rear of the machine (positive Y), or a counter-clockwise arc as viewed from the front of the machine. This is a move such as:
g21 g18 g2 i20 x40 f300    or
g21 g18 g2 i20 f300  (full circle)

I observe this to rotate in the CCW direction as viewed from the front. Is this your understanding? As I said, I could be misinterpreting things.

aldenhart, 2015-02-09:
Please pardon any interim chatter. I have been able to reproduce your issue.

aldenhart, 2015-02-10:
I have a fix in test. Should be able to push to edge in the next few days.

aldenhart, 2015-02-23:
This issue is fixed in edge build 440.13 (and above). Please let me know if you have any questions.

closed 2015-02-23 by aldenhart

thrushd, 2015-02-23:
Awesome, thanks!