synthetos/TinyG #121: PWM Outputs and Calibration

cmcgrath5035, 2015-01-15:
I am testing tinyGv8 with fw build 438.02.
I have buffered the pwm outoput (3.3v->5V) and am driving a PWM controller(the one Inventables sells).
That unit supposedly uses the pwm input to directly drive the FET driver, range 1KHz-10KHz.
I am monitoring the pwm output on tinyG with a USB oscope.
1. What I am seeing is that the pwm output frequency is 1/2 the $p1frq setting. I can't guarantee the scope accuracy, have see the same result with settings of 100Hz, 10000Hz and 20000Hz.  I am currently running with $p1frq=20000 and hear an annoying ~10kHz whine.
2. I find that a tinyG reset ( ctl+x or button) is required to complete the reset of $p1frq. Is this correct?

ril3y, 2015-01-15:
In my recent experience you have to reset the board to have the pwm
settings take affect.  ANY of them.

cmcgrath5035, 2015-01-16:
I was able to change phase settings, but have to add sometimes. Other attempts generated error messages or required two tries

In my particular case, there is an interesting interaction among $p1frq and the phase settings required to achieve full RPM and particularly in the slower speed rotation settings. More testing required.

Do you see the same 1/2 $p1frq operation?

ril3y, 2015-01-16:
Can you explain more?  Also give me your commands I will verify on my scope.

mikest, 2015-01-16:
is the PWM driven controller you're using setup for hobby RC timings or something else?

also, yes: you will need to reset after changing the PWM driver timing for the changes to take effect.

--mikest

cmcgrath5035, 2015-01-16:
This is the controller I am using:  http://www.amazon.com/RioRand-Driver-Controller-Spindle-Governor/dp/B00HUQY9HC/ref=pd_cp_hi_1
Also Available from Inventables for their Quiet Spindle.  
One Spec of interest:
**5.Except the board inside a PWM speed control, can also support an external PWM signal input speed control, external PWM input requirements: level 3.5-12V VPP, the frequency of 1K-10KHZ, in particular apply in MCH3 spindle speed control.**

Controller has an on-board 555 based pwm oscillator driven by pot or can bypass that and drive from external PWM. Controller seems to be working OK.

As to $p1frq setting, the compiled default with 438.02 is 100 Hz, but I was seeing 50Hz. I set it to $p1frq=10000, which did not do anything until I tried the reset, which did reset frequency, but to 5KHz.
I tried again, $p1frq=20000 , reset and get 10KHz.

This is measured at output of tinyG, buffered by a 74AHCT125 Quad 3.3V-5V bus driver. The PWM input to the controller is an opto isolator, Speced at 3.5V min so I decided not to mess with the tinyG 3.3V signal, although that is the way Inventables has folks using it (See:https://www.inventables.com/projects/quiet-cut-spindle-upgrade-tinyg} 
The controller actually does work with 50Hz input, but the motor growls a bit.

An item of note - I initially played with the setup by adjusting $p1pof  in 0.1 increments from 0 to 1 then issuing an M05, which seemed to re-read the value I had set. Used a Laser Tach to measure Spindle rpm, then came up with low and high settings.

cmcgrath5035, 2015-01-16:
Here is a bit more data
1. I am somewhat confidant in my Oscope now - it has a built in 1kHz calibration port that does measure 1kHz
2. I made a quick check on my tinyGv9, running G2_ Edge 71.04.
There $p1frq=100 in the default build.
It measures 100Hz.

So $p1frq issue is on the v7 (and v8?) platform?

cmcgrath5035, 2015-01-16:
OK, go back to your day jobs. 
I powered everything down last night, leaving $p1frq=15000, which (after reset) last night was giving a Oscope measurement of 7.5KHz.
This AM, powered up and the PWM output is now running at 15kHz. 
Maybe it takes two resets?

On powerup this AM

cmcgrath5035, 2015-01-16:
Is there a status request that would return what the PWM is currently doing.
For example, I issue a M03 S8000 command. The motor starts up.
My low freq is set to 5000, my high freq is set to 10000.
So tinyG is interpolating between low and high (linear, I assume) and choosing a PWM phase and setting it.
Can I query to find out what phase was determined?

mikest, 2015-01-16:
the scalar is mapped linearly, so if it matches the frequency you will get a 1:1 mapping.

--mikest

cmcgrath5035, 2015-01-17:
You might find this test summary informative https://dl.dropboxusercontent.com/u/50261731/Spindle%20Test%201.doc
My testing is ongoing, but these are likely better ShapeOko parameters than what are currently in the ShapeOko settings .h listings
Resetting (I used cntl+x)does help a lot with setting PWM parameters, but there still seems something funky in tinyG. Frequently, an attempt to modify a parameter, eg $p1cpl=0.1 will be rejected(error msg) but $p1cpl=0.100 will work. Other times, $p1cpl=0.200 will fail the first try, but works when simply reentered.

Repeat question: With the above settings, I issue M03 S8500 and the spindle spins up somewhere between low and high. Is the actual phase setting computed and used buy tinyG available via a status query?

mikest, 2015-01-18:
tinyg is "unresponsive" when writing to eeprom, if you're seeing successive writes fail you might be issuing them too fast. wait until the response comes back before issuing the next settings update.

for PWM, did you read how it works? the driver just takes one numerical range (the S-word) and uses it to map that to the phase in another frequency range. it's a linear mapping, so if your PWM controller is not linear you will get odd results. this is why there is a setting for the "off" PWM value, the initial start value may be a bit higher.

I primarily designed the PWM driver to control hobby RC brushless ESC controllers which have a 1-2ms pulse width range.

--mikest

cmcgrath5035, 2015-01-18:
Thanks Mike
Yes, I think I understand the Mapping process, my assumption (and what I 
think I see) is
Linear Mapping between p1csl and p1csh
So, if Gcode  M03 S/wxyz/ has
    wxyz < p1csl  -> phase = p1cpl
    p1csl<=wxyz<=p1csh  -> phase= linear mapping between p1cpl and p1cph
wxyz> p1csh  -> phase = p1cph
I can of course do that math, I was just asking if tinyG could tell me 
what value of phase is being fed to the PWM generator.
I realize that may not actually be a number between 0 and 1 internally.
I'll just assume that the answer is no, it cannot be read dynamically 
with a status.

I am not really as clear on why $p1pof is settable, or more importantly 
why it does not default to 0.000
With my controller, motor start up below phase=0.070 is unreliable, 
'never' happens above 0.060.
So I could set p1pof to 0.060, but I have no way of knowing if that will 
always not start the motor, particularly if i were to
manually spin the spindle such as doing ajob-active tool change during a 
pause in Gcode (CP makes this scenario very real).
I believe p1pof=0.1 in most of the default builds.
That would start my spindle on startup.

I realize there is very little standardization of PWM, controllers, 
typical motors and impossible to find specs.
so there very well could be a controller that needs p1pof to be >0.000 
for the motor to be really off.

Cinezaster, 2015-01-19:
Indeed their are little standardizations of PWM regarding controllers. 
I'm building a speed controller for an AC 100W 220V motor. (doing a roland camm-3 conversion)
I read the pwm duty to set the speed of the motor. In my case I can choose my settings.

cmcgrath5035, 2015-01-19:
Cinezaster - Please expand on "I read the pwm duty to set the speed....". You read it from where? I assume you have your own machine controller application, it reads PWM (from.....?) and then talks to your custom speed controller?

Cinezaster, 2015-01-19:
Right now I'm developing a speedcontroller with an Arduino Leonardo.
- reading the pulses of the spindle.
- reading the duty of the tinyG pwm output.
- reading the spindle on/off output of the tinyG.
- reading the mains zero-crossing.
- using PID control to set the phase control on a random triggered solid state relay.

I was first thinking of listening in on the serial connection checking for M3 M4 and M5 commands. But since these are buffered I went for this approach. I don't need the direction because a single phase AC motor has only one direction.

JimF42, 2015-02-02:
Cinezaster, are you planning on having your speed controller be open source or is this just personal. Also, I have been debating on going with a DC motor, but I can't find any power controllers that are closed-loop and will adjust the spindle speed as you are talking about.  Do you know of any DC versions?

Cinezaster, 2015-02-03:
Yes the project of the AC speed controller will be opensource http://cinezaster.github.io/Camm-3_2_tinyG/ Not everything I got is in their yet. But I'm gonna pick up the project next week, to bussy right now. 
I don't know any closed-loop controllers for DC version but that would not be so difficult to build as an AC controller because you would use direct pwm control.

cmcgrath5035, 2015-02-03:
JimF42 - Is this project http://www.truetex.com/dcdrv.htm  along the lines of what you are searching for?  Certainly 'industrial' scale, but may lead to a smaller scale version.

closed 2015-02-20 by aldenhart