synthetos/TinyG #124: Power Management bugs

JimF42, 2015-01-19:
I have found what appears to be two bugs related to power management.

I am running 438.02 build 0.97 and what I am experiencing from the documentation at the Power Management page https://github.com/synthetos/TinyG/wiki/Power-Management is inconsistent.

My biggest issue/bug is that using $*N*pm=2 is not keeping the motors energized. All my motors are setup with pm=2 and only the motor that is traveling is kept powered up beyond $mt. This is causing  my Y axis to become deenergized and while the spindle is cutting in the X direction, the spinning cutter is causing significant Y drift. The documentation states "This will turn on the motor power at the start of **_any move on any axis_** (a "cycle"), and will de-energize the motors MT seconds after the cycle is complete (all motion stops). The timeout interval is set by the MT value."

The other power management related issues are not as a big deal, but probably should be addressed. According to the same web page, $me=N should energize the motors for N seconds. For me, it does nothing (no-op). It also says that for JSON mode that {me:60} should also enable the motors for 60 seconds. In that case it also does nothing along with {“me”:60}. But, issuing {“me”:”60″} enables the motors for the default motor timeout value (mt).

Lastly, the documentation states that $*N*pm=0 prevents the motor from turning on, but in fact all it does is disable power management, including honoring the power timeout ($mt).

Thank you,
Jim

aldenhart, 2015-01-21:
Thanks for the heads up. I will take a look at this once I return from travel next week.

JimF42, 2015-01-21:
Thanks for looking into it. I wanted to add that I verified the primary issue (after my cut failure) by simply doing a long air cut in the X direction--one that took longer to cut than my current value of $mt. I was then able to freely push and my Y axis back and forth after the $mt.

Then, in my last paragraph, it appears I misspoke. I said "... but in fact all it does is disable power management, including **honoring** the power timeout", when in fact it **ignores** the power timeout. I am assuming that this is simply a documentation bug and TinyG is doing what it was designed for with pm=0.

Sorry for any confusion.

JimF42, 2015-01-21:
I took a look at the code (but I don't have a programmer/debugger, so I can't update my tinyg to test), but I am guessing that the following code in stepper.c may be the issue.

    ISR(TIMER_DDA_ISR_vect)
    {
    ... removed for brevity
    	if (--st.dda_ticks_downcount == 0) {		// end move
     		TIMER_DDA.CTRLA = STEP_TIMER_DISABLE;	// disable DDA timer
    		st_start_disable_motors_timer();
    		// power-down motors if this feature is enabled
    		if (cfg.m[MOTOR_1].power_mode == true) PORT_MOTOR_1_VPORT.OUT |= MOTOR_ENABLE_BIT_bm; // set to 0 to disable
    		if (cfg.m[MOTOR_2].power_mode == true) PORT_MOTOR_2_VPORT.OUT |= MOTOR_ENABLE_BIT_bm;
    		if (cfg.m[MOTOR_3].power_mode == true) PORT_MOTOR_3_VPORT.OUT |= MOTOR_ENABLE_BIT_bm;
    		if (cfg.m[MOTOR_4].power_mode == true) PORT_MOTOR_4_VPORT.OUT |= MOTOR_ENABLE_BIT_bm;
    		_load_move();							// load the next move
    	}
    }

Shouldn't `if (cfg.m[MOTOR_**n**].power_mode == true)...` be `if (cfg.m[MOTOR_**n**].power_mode == 3)`? It may be more complicated than that, but that just doesn't seem right to me. Otherwise you are powering down the motor if any mode other than zero is selected, right?

Hope my armchair coding is of some use to you.
Jim

aldenhart, 2015-01-26:
Thanks for the info. I've returned from travel and am digging into this.

aldenhart, 2015-02-10:
I've got a version in test that fixes this. Should push to edge some time this week.

aldenhart, 2015-02-23:
Please check out edge build 440.13 or later. Power management behaviors are documented here:
https://github.com/synthetos/TinyG/wiki/Power-Management

closed 2015-02-23 by aldenhart

JimF42, 2015-02-23:
Thanks for the update. I will try it out, but probably not before the weekend.

I have one question and one observation.

The question is, since I have never updated my TinyG firmware, will my settings be preserved, or do I need to backup my settings ($$, I think) before doing the update?

The observation is, on the wiki page you referred above, there are still two places where the behavior is not consistent with the sample. I assume the sample is wrong. The samples refer to motor locking *time* and not motor *number*.

> Text mode examples:
> $me=180    Lock all motors for 3 minutes for a tooling operation
> ...
> JSON mode equivalents:
> {me:180} 

Thank you,
Jim

ril3y, 2015-02-24:
Backup your settings yes.  They will not persist.