synthetos/TinyG #125: Velocity not working on edge build

smartrobot, 2015-02-02:
I have a shapeoko 2. I just switched from master to edge because I was not able to get Chilipeppr to working in 380.08, but when I switched to edge 438.02 all of my problems seemed to be fixed other than velocity stopped working. Is there a bug or has do I need to change my settings somewhere?

ril3y, 2015-02-02:
"velocity stopped working" is impossible.  If you move you have velocity
:).  Perhaps you can describe what you are experiencing?  My guess would be
your jerk settings for each axis.  $xjm $yjm etc..

Riley

smartrobot, 2015-02-02:
When I was on an earlier build the motors would slowly ramp up and then slow down as they are getting into to new position, but now they are constantly at full throttle.

ril3y, 2015-02-02:
This is almost 100% a jerk setting error.

What is your $xjm $yjm $zjm?  Try lowering it and see if it improves.

cmcgrath5035, 2015-02-02:
If I recall correctly, the jerk units changed between 380.08 and 438.02.
I run NEMA 17s with the following:
[xjm] x jerk maximum           5000 mm/min^3 * 1 million
[yjm] y jerk maximum           5000 mm/min^3 * 1 million
[zjm] z jerk maximum             50 mm/min^3 * 1 million

Not necessarily the best values for your setup, but something to compare to

JimF42, 2015-02-02:
So my Shapeoko 2 is **not** supposed "jerk" when it does full motion (G0) moves? I have setup my jerk settings according to the TinyG Shapeoko setup page (https://github.com/synthetos/TinyG/wiki/TinyG-Shapeoko-Setup) and that is exactly what my machine does. If I position rapidly in X or Y (G0) to any where over say a inch or so away my machine lurches a small amount.  

I am not at a place where I can double check my numbers, but they *should* be the same as cmcgrath5035's.

I am running what was already preinstalled, 438.02 build 0.97.

cmcgrath5035, 2015-02-02:
If I guess a bit what your definition of 'lurch' is, I'd suggest reducing  $XVM and $YVM to something less than 16000 mm/min, which is likely where you are at if you follow the Synthetos suggestions. When I had a DW660 (large, heavy) spindle attached. that mass made the machine sort of rattle (lurch) if moves were long enough to actually approach the Velocity maximum. I have since converted to a lower mass PWM spindle. Unless your jobs have a lot of G0 moves, lower VMs don't add much time. I usually cut at or below F800, so lurch during G1, G2 and G3 moves was not an issue. Unless the perceived lurch is causing your machine to become inaccurate (e.g. slip on the belt), it is really user preference tuning.

aldenhart, 2015-02-02:
Carl's suggestions are good. The other thing you can try is adjusting the maximum jerk in the axis settings, $XJM and $YJM. I actually have my Shapeoko2 running at 40,000 mm/min rapids now that it's tuned up. The default in the profile of 16,000 mm/min is a conservative value that should work on most setups, but it can do a lot more than that if your belts are tight and the machine is in good square. I don't mind a bit of "jump" in the rapids, and I am not moving a massive spindle.

JimF42, 2015-02-02:
I see... so the values that I need to specify for jerk motion control also need to take into account the mass of the spindle too (how am I to know or calculate that?). It is interesting that the values are so high then in the TinyG Shapeoko Setup page as the "default" spindle is a Dremel-like tool.  I in fact, am using a Dremel brand tool as I heard too many negative stories about the knock-off tool that comes with the standard package.

That said, I was originally expecting a description more like in the original post where the machine may go full speed, if it can, but it would ramp up and down the speeds during it's traversal. Is there a way of tuning TinyG to make _that_ happen? Or, is that what xjm/yjm sets and not just the max traversal speed?

cmcgrath5035, 2015-02-02:
When you ask "how am I to know or calculate that?", the simple answer might be either get an A+ in "MechE403 - Complex Machine Dynamics", or do it the hobby CNC way, experiment! 
Depends on what you are calling Full Speed. In Paramater terms, $xmv and $Ymv are "Full Speed for G0 moves". The Fxxx setting in your G code file normally sets a much slower speed for machining moves, G1, G2, and G3. If your Gcode does not set an F value, then G1, G2 and G3 go at $XVM and $YVM. I highly recommend not trying to use a spindle to machine something at 16000 mm/min. But you could have a laser instead of a spindle;  you understand the difference?
Think of jerk as a parameter that quantifies the aggressiveness with which the motors will attempt to change speed, such as ramp up then ramp down to make a traverseas you ask. GRBL uses linear equation ramps, tinyG implements third order equation ramps. 
Also, tinyG is always planning for the next move by looking ahead at next lines of Gcode. I knows that making a 160deg left turn will potentially create a lot mor "lurch" than making a 10deg left turn, as a simple linear move example.
That help?

JimF42, 2015-02-02:
@cmcgrath5035,

Yes, I think this helps. So, in short, $xmv and $ymv say what my top speed is and $xjm and $yjm (jerk settings) set the ramp up and ramp down speed curve. So if I want a less aggressive ramping up and down (thus less lurching of my machine), I would lower the jerk values until I see the response I am looking for. And how "smooth" this motion is will also depend on the weigh of my spindle (in my case on the lighter side).

If this is correct, could _someone_ update the Shapeoko Setup wiki page with this extra detail? When I read the Shapeoko setup page, I get the _impression_ (at least with the jerk settings) that this is what the value needs to be. Helping others understand that this value may be great for starting with, but can be tuned like some of the other values, would be helpful. I can change the page, but I am so new to this I am not sure if I am qualified (may state something wrong for example...).

smartrobot, 2015-02-02:
I just want to say thank you for all of your help and I was able to fix my problem.

closed 2015-02-02 by smartrobot

cmcgrath5035, 2015-02-02:
I'll give your first paragraph a 90% conceptual grasp, and10% yes-but.
Remember, there is only one jerk parameter, applied to both G0 Moves, 
governed by VM parameter, and G1,G2 and G3 moves.
governed by Fxxx setting. For most productive machining (G1, G2 and G3), 
you want optimal jerk for F speed, then tune your VM parameters for 
tolerable lurch. So VM, F and jerk interact a lot.

And, keep in mind Alden's comment that his ShapeOko is more aggressive 
then the values we are discussing.
Are you sure you machine is assembled correctly and parameters are correct?
Maybe have a look here 
https://www.synthetos.com/topics/forum-information-sharing-readme/
and post your full parameter list.
I am assuming a ShapeOko2 (4 motors).
Perhaps your Y and Yr motors are not in perfect harmony?

I'll have a look at ShapeOko, but suggest you spend some quality time in 
the TinyG Wiki as well, to get to know what is there.
https://github.com/synthetos/TinyG/wiki

And discussions like this are probably better in the tinyG Support Forum
https://www.synthetos.com/forums/tinyg/
This Issues list is really for issues that might be bugs, but there are 
no hard bounds on the subjects.

ril3y, 2015-02-02:
jim.

Just to correct for others that might read this XVM and YVM not XMV YMV.

Good to hear you got it.

cmcgrath5035, 2015-02-02:
OOps, sorry. I have been debugging some JS for Chilippr all day - eyes 
and fingers failing me.

Good catch, Ril3y.