synthetos/TinyG #126: Configuration Management

cmcgrath5035, 2015-02-07:
Is there a JSON equivalent  to text = "$$" ?

Hint: Working on Backup/Restore in CP, for now in Text mode.
Would try JSON, but don't see a way to "learn" what the full command set is, with descriptive info, as $$ delivers in text mode.

aldenhart, 2015-02-07:
The best way to do this is to process a list of objects on at a time, such as:
{sys:n}
{1:n}
{2:n}
{3:n}
{4:n}
{x:n}
{y:n}
{z:n}
{a:n}
{b:n}
{c:n}

and if you need them:
{p1:n}
{g54:n}
{g55:n}
{g56:n}
{g57:n}
{g58:n}
{g59:n}

{g28:n}
{g30:n}

cmcgrath5035, 2015-02-07:
Thanks, understand.

My objective is to produce a Backup up as would be made the old fashion way
1. $$ in Terminal window, scrape the console output and save as text file
2. A side benefit is that one gets a listing of ALL settable parameters plus those Synthetos decides might be useful for the user to see (RO Parameters such as $fb, $fv, etc)
3. And a third, as things change, is that $$ Self-Updates as new parameters are added.
4. And a fourth - the terse descriptive text is displayed as well.

I have a completed Alpha version of text mode based backup/restore, wuill send you guys a link to play with after a bit more clean-up. Will run as a button in the TinyG widget at CP, most likely.

Having slugged thru the development, a.k.a. learn JS one line at a time (plus of course, steal from others), I can see benefits of doing more of it in JSON, but feel that the loss of some of the "old-way" features would not be as helpful, particularly for new starters.

And while I have your attention:
In combing thru the tinyG2 .h files on Github I see reference to user parameters. Are they implemented yet?
What command(s) access them.  They might be helpful for CP widgets wanting to do "save last state in CP" type info.

ril3y, 2026-11-16:
There is no single JSON dump; read each group (`{sys:n}`, `{1:n}`..`{4:n}`, `{x:n}`..`{c:n}`, `{p1:n}`, `{g54:n}`..`{g59:n}`). Text-mode `$$` remains the full human-readable listing.

closed 2026-11-16 by ril3y