synthetos/TinyG #127: Message when motors are powered off

jkuusama, 2015-02-20:
My machine would benefit knowing if the motor power timer has tripped (and therefore, the machine lost its accurate position and needs to recalibrate itself).

aldenhart, 2015-02-20:
This is in the works and in the current edge build 439.06 - but it's still in testing. The command is:
$pwr1... or just $pwr for all 4
{pwr1:n} or {pwr:n}

It reports the enabled state of the motors.

There will be a new edge push this weekend that addresses some other issues, so you might want to wait for that.

jkuusama, 2015-02-20:
Cool, thanks!

-Juha

jkuusama, 2015-02-21:
Is there a message that goes out without asking for the status?

aldenhart, 2015-02-23:
Juha,

You can add the power indicators to your status report and it will be returned. I recommend using filtered status reports (as always).

Add this to your report setup string:

“pwr1”,“pwr2”,“pwr3”,“pwr4”,

…or whatever motors you want. Use edge build 440.13 or later.

Alden

closed 2015-02-23 by aldenhart

jkuusama, 2015-02-23:
Thank you! I’ll try this out this week.

 

-Juha