synthetos/TinyG #129: G91 not recognized by gcode module, but is recognized by serial port console?

secretspy711, 2015-02-22:
I'm running the gcode file below in the jarret workspace.  When the gcode module reaches the G91 command (I've highlighted it with asterisks below), the machine stops, that line turns pink indicating "error/unsupported command" and requires a ctrl+X.  However, when issuing the same G91 via the serial port console, it behaves as expected.

( Made with CamBam )
( acrylic_cover)
( 2/18/2015 8:48:49 PM )
( T0 : 0.135 )
G20
G90
G64
G40
G0 Z0.2
( T0 : 0.135 )
T0 M6
( Drill1 )
G17
M3 S1000
G0 X0.7475 Y4.1795
G0 Z0.2
G0 X0.76 Y4.1795
G0 Z0.1
G1 F3.0 Z0.0
G2 F15.7 X0.7412 Y4.1687 Z-0.0267 I-0.0125 J0.0
G2 Y4.1903 Z-0.0533 I0.0062 J0.0108
G2 X0.7582 Y4.186 Z-0.073 I0.0063 J-0.0108
G2 X0.7478 Y4.167 I-0.0107 J-0.0065
G2 X0.7365 Y4.1855 I-0.0003 J0.0125
G2 X0.7582 Y4.186 I0.011 J-0.006
( go to specific position.nc )
G91 (relative coordinates *************************)
G0 Z2 (go up 2 inches)
g90 (absolute)
G0 X-2 Y12 (get out of the way to certain spot)
M0 (pause)
G0 X2 Y2 (go here before lowering bit)
G90 (make sure it's in absolute mode)
( Profile1 )
S1000
G0 Z0.2
G0 X1.33 Y4.1795
G0 Z0.107
G1 F3.0 Z-0.073
G2 F15.7 X0.4562 Y3.675 I-0.5825 J0.0
G2 Y4.684 I0.2912 J0.5045
G2 X1.33 Y4.1795 I0.2912 J-0.5045
( Profile2 )
S1000
G0 Z0.2
G0 X0.2062 Y-0.0675
G0 Z0.107
G1 F3.0 Z-0.073
G1 F15.7 X1.2887
G3 X1.3562 Y0.0 I0.0 J0.0675
G1 Y0.2545
G1 Y0.2845 Z-0.013
G1 Y0.4195
G1 Y0.4495 Z-0.073
G1 Y1.9245
G1 X1.3365 Y1.9443
G1 X1.3562 Y1.9245
G1 X1.495
G3 X1.5625 Y1.992 I0.0 J0.0675
G1 Y2.4133
G1 Y2.4433 Z-0.013
G1 Y2.5783
G1 Y2.6083 Z-0.073
G1 Y4.6496
G1 Y4.6796 Z-0.013
G1 Y4.8146
G1 Y4.8446 Z-0.073
G1 Y4.942
G3 X1.495 Y5.0095 I-0.0675 J0.0
G1 X0.0
G3 X-0.0675 Y4.942 I0.0 J-0.0675
G1 Y4.8618
G1 Y4.8318 Z-0.013
G1 Y4.6968
G1 Y4.6668 Z-0.073
G1 Y2.5825
G1 Y2.5525 Z-0.013
G1 Y2.4175
G1 Y2.3875 Z-0.073
G1 Y1.992
G3 X0.0 Y1.9245 I0.0675 J0.0
G1 X0.1387
G1 X0.1585 Y1.9443
G1 X0.1387 Y1.9245
G1 Y0.4237
G1 Y0.3937 Z-0.013
G1 Y0.2587
G1 Y0.2287 Z-0.073
G1 Y0.0
G3 X0.2062 Y-0.0675 I0.0675 J0.0
G0 Z0.2
M5
M30

ril3y, 2015-02-22:
Why are you running a grbl workspace with a tinyg? I would re run this at
the correct chilipeppr workspace. /tinyg not Jarret. Try that first.

secretspy711, 2015-02-22:
Perhaps I posted this in the wrong area.  I am running a grblshield.

ril3y, 2015-02-23:
Ahh yes you have.  You should contact the Grbl guys.  These are not the
droids you are looking for :)  We make the TinyG code base, not the grbl
code base :)

Ril3y

closed 2015-02-23 by aldenhart