synthetos/TinyG #130: PWM related to speed

stg, 2015-03-02:
In a CNC laser cutter (and to some extent CNC mills) it can be quite important to have the capability to scale the PWM (laser power/spindle speed) according to the current (calculated) feed speed.

For example, if you want to cut a straight line on a laser cutter at 100% power, the power needs to be ramped up as the head accelerates and ramped down as the head decelerates or the cut can result in burns at the edges of the lines. This becomes even more important if you are trying to engrave lines (non-raster) on the material - the material will have burns at every sharp bend of the design.

For a CNC mill, working with thermoplastics, it can be useful to have the motor speed ramp down to prevent melting of the plastic in detailed patterns resulting in slower cutting speed due to frequent sharp bends.

This also needs a configurable lower bound (for example "5%") to account for the fact that a laser tube (and motor, depending on controller) will typically output nothing below a certain threshold. So a 0-100% feed speed (where 100% is the currently configured feed rate) would be mapped to "5%"-100% output power (where 100% is the currently configured power), continuously as the motion controller works it's magic.

Setting this lower bound to 100% would effectively disable the feature.

I'm guessing this may be code-wise related to the "override" feature for spindle speed (PWM) and feed rates that have already been requested and may need to be planned in conjunction, if accepted.

giseburt, 2015-03-02:
This is a great idea, and something we actually have planned to work on. Unfortunately, I couldn't say when at this point.

Thank you!
  -Rob

stg, 2015-03-02:
Seeing as I'm quite excited about, and impressed with, this project as well as having a dire need to replace the hardware and software of both my CNC mill and laser cutter - I am truly hoping to find some time to contribute as soon as I've finished building the cooling- and chip-/gas-evac- -systems for said machines. Picking up a bunch of Due's this week to start fidgeting with G2, hoping to port it to a faster Cortex-M4 and look into breaking out stepper.c into a small FPGA or CPLD before digging into the rest of the code. I'd very much like to be able to run that final stage perfectly jitter free, at MHz rates aaand there I went all off topic again...

openhardwarecoza, 2015-07-14:
Positively confirming my need for this as well.    The lasersaur guys has it in the Grbl fork and it makes a huge difference

aldenhart, 2015-07-14:
Yes. This is in the works. Thanks for your patience.

myk3y, 2019-04-30:
Four years on - any progress?