synthetos/TinyG #131: 440.14 not playing well with Chilipeppr

JimF42, 2015-03-13:
I upgraded my TinyG to 440.14 and now when I cut Chilipeppr is not working correctly. I first opened a thread in on ChiliPeppr, but they say it must be a firmware bug and that I need to open the ticket here...

(original Chilipeppr ticket) https://github.com/chilipeppr/tinyg/issues/20

Are you aware of any issues with 440.14 that would cause Chilipeppr to not cut correctly? As outlined in the Chilipeppr ticket, the simulation works perfectly, but when I go and cut, after every step pass the machine goes back to the XY origin, does a funny "dance" (for the lack of a better term), then drags the tool back to do the next step-down pass--all without resetting the Z to a safe height and dragging the end mill through the material.

1. Where there any changes to the status codes (perhaps) with 440.14? I ask because while the simulation runs as expected, the Chilipeppr GUI matches the movements of the CNC when trying to do an actual cut.
2. Is there anyway I can get the prior edge build .hex file? I only see 440.14 now.

The CAD was created in QCAD and the g-code with Cut2D. I will be more than happy to send files if it will help!

Thanks

aldenhart, 2015-03-13:
Yes, please post the file somewhere. Also, what build were you running before? -- Thx

JimF42, 2015-03-13:
I was running 438.02 build 0.97 before, and I just realized that I may not be running 440.14, but 440.13. I don't have the machine in front of me to verify. I will be happy to update and retry...

Let me know if you need more than the g-code file, which can be found at https://app.box.com/s/cgggzvphxlpxqz40obtj9p8rergztl6d.

aldenhart, 2015-03-13:
Yes, please retry with 440.14 and let me know.

JimF42, 2015-03-13:
I will try it this evening. Sorry for the confusion...

aldenhart, 2015-03-13:
No problem. Box is not letting me get that file for some reason. Is there some other place you could leave it? I'll be around this evening a few hours from now.

JimF42, 2015-03-13:
Let's try Dropbox then... 
https://www.dropbox.com/s/bj5kqi112gsdndp/Knob.gcode?dl=0

aldenhart, 2015-03-13:
I was able to run this file correctly with 440.14. Give it a try. There is a bug in 440.13 involving arcs. That may be what you saw.

ril3y, 2015-03-13:
I can confirm that 440.14 is working.  Yah go ahead and try it with 440.14.  If there IS still an issue please do the following.

$$

capture the whole output and paste it back here.  However, I think this is a non-issue in 440.14.

Thanks for the report though Jim.  We really take issues with our customer/code users, like this very seriously!

JimF42, 2015-03-14:
Yup, that was it. I didn't realize .14 came out so I hadn't flashed it on yet--I was still running 440.13.  Is there something I can subscribe to be notified when new edge updates are out? 

Thanks

ril3y, 2015-03-14:
Great.. I am going to close this issue.  Thanks for working with his though this.

closed 2015-03-14 by ril3y

JimF42, 2015-03-14:
@ril3y, the addition of the **Required** update notification sounds great!