synthetos/TinyG #133: Invalid JSON response for "gc"

Taxi4110, 2015-06-02:
If I send a gcode command and have set "gc":true in the status report setting, I get an invalid JSON response from tinyg!

SEND: {"gc":"g0y-10"}

I get this response:

RESPONSE: {"sr":{"line":0,"posx":0.000,"posy":-0.001,"posz":0.000,"vel":11.09,"unit":1,"stat":5,"gc":"{"gc":"g0y-10"}"}}

If I just want a status report without sending a g command, I get this:

{"sr":{"line":0,"posx":0.000,"posy":0.000,"posz":0.000,"vel":0.00,"unit":1,"stat":1,"gc":"{"sr":{"line":true,"posx":true, "posy":true,"posz":true, "vel":true, "unit":true, "stat":true, "gc""}}

Looks wrong to me... :)

aldenhart, 2015-06-02:
Thanks for reporting this. I'll find out what's up.

cmcgrath5035, 2015-06-02:
OK, now I am curious. I don't see "gc" listed in the table: https://github.com/synthetos/TinyG/wiki/Status-Reports#status-report-contents.
What is "gc"?

Taxi4110, 2015-06-02:
Gc stands for the gcode that was sent. I added this field in the status report because the wiki says:

Additionally, any valid token may be listed in a status report. For example, "g54x" will return the X offset in the G54 coordinate system (coordinate system #1). "fv" would return the firmware version.

giseburt, 2015-06-02:
I find this rather amusing. I don't believe we really wanted `gc` to be in a status report. (I could be wrong.)

However, it shouldn't be doing this nasty thing it's doing either.

Other than seeing if it'll work, is there a reason to put `gc` in your status reports?

Taxi4110, 2015-06-02:
My tought was, if the gcode file has no line numbers... so you can see what command is tinyg doing atm.
But I can add line numbers before passing the gcode to tinyg so I don't necessarily need this feature...

cmcgrath5035, 2015-06-03:
OK, understand what you are saying. But G54X is a parameter (and it's value returned), not Gcode oer se.

Not important what I think - I watch for Rob's more experienced answer.

Taxi4110, 2015-06-03:
I see.

When I set gc:true in the status report config, pass a gcode command to tinyg and get the right value back (current behavior atm), not just in a valid syntax, I'd like to think that this is just some kind of json formatting bug. :)

Like I said before... I don't need these feature necessarily, but it's nice to have. :)

ril3y, 2021-04-10:
Closing due to inactivity.

closed 2021-04-10 by ril3y