synthetos/TinyG #135: Probe report no longer in absolute coordinates

DanalEstes, 2015-06-25:
G38.2 results in a probe report such as:

{"r":{"prb":{"e":1,"x":0.000,"y":0.000,"z":0.259,"a":0.000,"b":0.000,"c":0.000}},"f":[1,0,0,9523]}

Coordinates in this report should be G53 absolute machine coordinates.  

Experimental verification shows that issuing G10 L2 and/or G28.3 commands will change the results, even when the physical touch occurs in the same location.  In other words, the results are affected by the current coordinate system.  

I believe this bug was introduced in the area of line 240, in module TinyG/firmware/tinyg/cycle_probing.c, at change: d3a78d1033568766f26369a5b4e9f74e07285db9

Which changed:
```javascript
		for( uint8_t axis=0; axis<AXES; axis++ )
		cm.probe_results[axis] = cm_get_absolute_position(ACTIVE_MODEL, axis);

	// if we got here because of a feed hold we need to keep the model position correct
	cm_update_model_position_from_runtime();
```
to:

```javascript
	for( uint8_t axis=0; axis<AXES; axis++ ) {
		// if we got here because of a feed hold we need to keep the model position correct
		cm_set_position(axis, cm_get_work_position(RUNTIME, axis));

		// store the probe results
		cm.probe_results[axis] = cm_get_absolute_position(ACTIVE_MODEL, axis);
	}
```

It appears that, post this change, the loop resets the axis positions to the runtime work coordinate system before it builds the results array.  The old code built the results first, while the system is still in absolute mode.   This change of order in the loop is somewhat hard to detect looking at a standard git diff page because braces were introduced; the reversal of sequence is quite clear when you view the files standalone pre/post change (or the code snippet above). 

Am I reading this correctly? And am I correct that prb reports should be absolute?

Thanks!

Danal

DanalEstes, 2015-06-26:
Fixed. Pull request is #136

DanalEstes, 2015-07-02:
I created a pull request and therefore closed this... I think that caused it to go un-noticed.  Therefore, reopening.  

This is fixed in pull #136

DanalEstes, 2015-07-02:
Clarified title.

aldenhart, 2015-07-04:
OK. Noted as well.

giseburt, 2015-07-09:
Digging into this now.

giseburt, 2015-07-09:
Ok, the meat of this problem is that at the end of the probe cycle it calls a function that returns the values in "units"  (inches or mm), and passes those results to a function that assumes mm (which all internal values are stored in). I believe that it needs to use a different (very similarly named) function that simply returns the values in mm.

I'm not sure that that there's not more to it at the moment. Once I am sure I'll put in a fix.

Thank you @DanalEstes for finding this and digging in enough to attempt a fix. I'm sorry I can't pull it in as-is.

DanalEstes, 2015-07-09:
No problem!  Absolutely no expectation on my part that my very first effort
would be pulled as-is.   :-)

ril3y, 2026-11-06:
Fixed in 450.03 (tag on 47cc6036): the probe report (`{"prb":{...}}`, `prbx`..`prbc`) now gives the machine position in mm, whatever G5x offset or G20 is active, as 440.x and g2core do. Test: tests/hil/tests/bugs/test_e028_probe_report_machine_mm.py. This is a change from 450.02, which reported work coordinates in current units.

closed 2026-11-06 by ril3y