synthetos/TinyG #138: Parameters not updating without Reset - tinyG V8

cmcgrath5035, 2015-07-19:
We are getting many reports from users having difficulty setting parameters to new values and making them usable - experience is, for example changing $1tr = 33  to $1tr = 40 is only effective after a tinyG reset. 
This is not my legacy experience and not the intended behavior, if I understand correctly.
Seems observed 440.15 and 440.16 versions, maybe 440.14.
Reports on both Forum and Chilipeppr Group.
I believe also seen from CoolTerm CLI.

Here is one thread: https://groups.google.com/d/msg/chilipeppr/IsKsirnIl2U/SwIYCGEYiyMJ

aldenhart, 2015-07-19:
OK. I've been able to reproduce this issue. Let me see what's up here and report back. FYI: I set up a bench (just a board and motors) and see the effect when changing $1tr. Are you aware of any other parameters people are aware of that are not changing in "real time"? I've tried and verified that this affects a $tr and $sa, but do you know if there are others?

cmcgrath5035, 2015-07-19:
Nope, Only $1tr and $sa that I have seen reported.
Don't have a tinyG with me to test with.

cmcgrath5035, 2015-07-21:
Looks like perhaps $ma as well is affected.
https://groups.google.com/d/msg/chilipeppr/oWrCHzFFUw4/qbXzvvDWafEJ

rfraties, 2015-07-27:
I have confirmed this problem.  I would like to also emphasize that a power-down / power-up sequence is NOT sufficient.  After having sent your parameters (using CoolTerm), then you must actually hit the reset button.  

I purchased the board around 3 months ago.  I was having intermittent issues with runaway cutting using CoolTerm.  Like 1000 cut commands would be fine and then you would get one random cut where it would go all the way across the workspace even past the hard limit, and cutting a big ugly line into your project.  So I decided to upgrade firmware to 440.16.  

The "easy" firmware upgrade did not work.  Had to use avrdude.  Had to include the avrdude "-e" parameter or it would not work.  But ultimately it did update.  

Then found that the z and y axis were moving much too fast, and x axis much too slow.  Checked the $tr settings and found that they had reverted to certain defaults which were wrong.  

Tried to change the settings but the settings would not actually apply.  Echoing the setting would give the correct new value, but the new value would not actually transfer to the motors.  For some of the settings, I think a power-down / power-up sequence did work.  But for the $tr settings, it did not work.  Finally found that hitting the reset button was the trick to get the $tr settings to apply.  

Now that the $tr commands have "taken" I will continue with my testing.  Hopefully the runaway problem is also fixed.  It is not always clear where the source of the runaway problem would be.  With CoolTerm or with tinyg?  Probably need to look for a CoolTerm alternative, just so I can rule it out as a possible source of problems.  I have checked and double checked my cut file and the cuts are correct.  There are no commands given to move outside the cutting area.  These runaway cuts, it is a real problem but hopefully it will now be fixed.  If not, I will put up another post.  

Thank you,

Roy

ril3y, 2015-07-28:
440.17 has been pushed to master.  It fixes the reset problem you described
above.  However, the tgupdater or "easy update" is not working in windows
right now. I am working on getting this fixed.  However in the meantime you
have to use avrdude as you discovered.

Sorry

Riley

ril3y, 2015-07-28:
The firmware is available here:

https://raw.githubusercontent.com/synthetos/synthetos.github.io/master/binaries/tinyg-master-440.17.hex

Right click  and save as the link above.  If you do not you run the risk of
windows browsers corrupting your hex file.

Riley

cmcgrath5035, 2015-07-28:
Here is an avrdude based procedure for Windows users : https://groups.google.com/d/msg/devtinyg/B9xSJaupzNQ/NDOABfAeheIJ

ril3y, 2026-11-06:
Fixed on edge-next and included in 450.03 (tag on 47cc6036): `$Ntr`, `$Nsa`, `$Nmi` and `$Nma` take effect without a reset. `$Nma` was the last one: a remap now resyncs the steppers (#294). A saved setting sent while a job runs is refused with status 117 rather than half applied.

closed 2026-11-06 by ril3y