synthetos/TinyG #140: Arc command (G2 & G3) first status report is wrong

cncgoko, 2015-09-18:
Hello,

I've been using TinyG for a while now and I recently switched to fb 440.18
Some strange behavior showed up.

When sending a G2 or G3 command, the first {"sr": ... } always shows the target point.
Is that on purpose ?

Below is a copy of the serial dialog

```
g0x0y0z0
{"r":{"gc":"G0X0Y0Z0"},"f":[1,0,9,734]}
 (  ...  move to origin )
{"sr":{"posx":0.000,"posy":0.000,"posz":0.000,"posa":0.000,"feed":1000.00,"vel":0.00,"unit":1,"coor":1,"dist":0,"frmo":1,"stat":3}}
{"qr":32,"qi":0,"qo":1}

( sending arc command )
G2 x0 y10 i5 j5 f1000
{"r":{"gc":"G2X0Y10I5J5F1000"},"f":[1,0,22,3845]}
 (Begin wrong status report )
{"sr":{"posx":0.000,"posy":10.000,"posz":0.000,"posa":0.000,"feed":1000.00,"vel":0.00,"unit":1,"coor":1,"dist":0,"frmo":1,"stat":3}}  
 (End wrong status report )
{"sr":{"posx":-0.530,"posy":0.593,"posz":-0.000,"posa":0.000,"feed":1000.00,"vel":26.61,"unit":1,"coor":1,"dist":0,"frmo":1,"stat":5}}
{"sr":{"posx":-1.256,"posy":1.724,"posz":-0.000,"posa":0.000,"feed":1000.00,"vel":1000.00,"unit":1,"coor":1,"dist":0,"frmo":1,"stat":5}}
{"qr":21,"qi":14,"qo":3}
{"sr":{"posx":-1.850,"posy":3.274,"posz":-0.000,"posa":0.000,"feed":1000.00,"vel":1000.00,"unit":1,"coor":1,"dist":0,"frmo":1,"stat":5}}
{"sr":{"posx":-2.067,"posy":4.921,"posz":-0.000,"posa":0.000,"feed":1000.00,"vel":1000.00,"unit":1,"coor":1,"dist":0,"frmo":1,"stat":5}}
{"qr":26,"qi":0,"qo":5}
{"sr":{"posx":-1.907,"posy":6.495,"posz":-0.000,"posa":0.000,"feed":1000.00,"vel":1000.00,"unit":1,"coor":1,"dist":0,"frmo":1,"stat":5}}
{"sr":{"posx":-1.371,"posy":8.068,"posz":-0.000,"posa":0.000,"feed":1000.00,"vel":1000.00,"unit":1,"coor":1,"dist":0,"frmo":1,"stat":5}}
{"qr":31,"qi":0,"qo":5}
{"sr":{"posx":-0.528,"posy":9.409,"posz":-0.000,"posa":0.000,"feed":1000.00,"vel":1000.00,"unit":1,"coor":1,"dist":0,"frmo":1,"stat":5}}
{"sr":{"posx":0.000,"posy":10.000,"posz":0.000,"posa":0.000,"feed":1000.00,"vel":0.00,"unit":1,"coor":1,"dist":0,"frmo":1,"stat":3}}
```

cncgoko, 2016-03-30:
Any update on this issue ?

aldenhart, 2016-03-31:
I am working on a bunch of issues in edge-0.97. This is on the list. Please watch for updates.

ril3y, 2026-11-16:
Fixed in 450.01: the first status report after a G2/G3 now shows the start of the arc, not its endpoint.

Fix: https://github.com/synthetos/TinyG/commit/16aef83f (first released in [450.01](https://github.com/synthetos/TinyG/releases/tag/450.01))
Regression test: `tests/hil/tests/bugs/test_e090_arc_endpoint_in_sr.py`

closed 2026-11-16 by ril3y