synthetos/TinyG #141: G3 command make arc + 1 full revolution and wrong direction

cncgoko, 2015-09-20:
Hello,

Firmware build : 440.18
Issuing a G3 command on the G18 plane move TinyG 1 full revolution + the expected arc move. My guess is that it turns in the wrong direction (In G18 plane, both G2 and G3 turns in the same direction).  On G17 and G19 they work properly (G2 works properly in all planes).

                |       G2     |       G3
------------ | ------------ | -------------
G17         |     OK       |      OK               
G18         |     OK       |      **BUG**
G19         |     OK       |    OK                   

For instance :
```javascript
g0 x0 y0 z0
g17 g3 y20 j10   // works properly
```

```javascript
g0 x0 y0 z0
g19 g3 z20 k10   // works properly
```

```javascript
g0 x0 y0 z0
g18 g3 z20 k10   // bug
```
Below is an extract of the communication from TinyG. We can see that X position and Z position loop.
```
G0 X32.5 Y11.625 Z20.
	{"r":{"gc":"G0X32.5Y11.625Z20."},"f":[1,0,22,566]}
	{"sr":{"posx":-51.032,"posy":11.625,"posz":15.000,"posa":0.000,"feed":500.00,"vel":15.64,"unit":1,"coor":1,"dist":0,"frmo":1,"stat":5}}
	{"qr":31,"qi":1,"qo":0}
	{"sr":{"posx":32.500,"posy":11.625,"posz":20.000,"posa":0.000,"feed":500.00,"vel":0.00,"unit":1,"coor":1,"dist":0,"frmo":1,"stat":3}}
	{"qr":32,"qi":0,"qo":1}
G18 G3 X27.5 Z15. I-5. K0. F500.
	{"r":{"gc":"G18G3X27.5Z15.I-5.K0.F500."},"f":[1,0,33,7997]}
	{"sr":{"posx":27.500,"posy":11.625,"posz":15.000,"posa":0.000,"feed":500.00,"vel":0.00,"unit":1,"coor":1,"dist":0,"frmo":1,"stat":3}}
	{"sr":{"posx":32.498,"posy":11.625,"posz":19.966,"posa":0.000,"feed":500.00,"vel":64.45,"unit":1,"coor":1,"dist":0,"frmo":1,"stat":5}}
	{"sr":{"posx":32.484,"posy":11.625,"posz":19.754,"posa":0.000,"feed":500.00,"vel":168.62,"unit":1,"coor":1,"dist":0,"frmo":1,"stat":5}}
	{"sr":{"posx":32.467,"posy":11.625,"posz":19.486,"posa":0.000,"feed":500.00,"vel":138.04,"unit":1,"coor":1,"dist":0,"frmo":1,"stat":5}}
	{"sr":{"posx":32.460,"posy":11.625,"posz":19.371,"posa":0.000,"feed":500.00,"vel":14.85,"unit":1,"coor":1,"dist":0,"frmo":1,"stat":5}}
	{"qr":4,"qi":29,"qo":1}
	{"sr":{"posx":32.459,"posy":11.625,"posz":19.366,"posa":0.000,"feed":500.00,"vel":10.18,"unit":1,"coor":1,"dist":0,"frmo":1,"stat":5}}
	{"sr":{"posx":32.439,"posy":11.625,"posz":19.260,"posa":0.000,"feed":500.00,"vel":141.19,"unit":1,"coor":1,"dist":0,"frmo":1,"stat":5}}
	{"sr":{"posx":32.370,"posy":11.625,"posz":18.901,"posa":0.000,"feed":500.00,"vel":267.92,"unit":1,"coor":1,"dist":0,"frmo":1,"stat":5}}
	{"qr":3,"qi":2,"qo":1}
	{"sr":{"posx":32.219,"posy":11.625,"posz":18.376,"posa":0.000,"feed":500.00,"vel":491.95,"unit":1,"coor":1,"dist":0,"frmo":1,"stat":5}}
	{"sr":{"posx":31.889,"posy":11.625,"posz":17.609,"posa":0.000,"feed":500.00,"vel":500.00,"unit":1,"coor":1,"dist":0,"frmo":1,"stat":5}}
	{"qr":4,"qi":2,"qo":3}
	{"sr":{"posx":31.454,"posy":11.625,"posz":16.950,"posa":0.000,"feed":500.00,"vel":500.00,"unit":1,"coor":1,"dist":0,"frmo":1,"stat":5}}
	{"sr":{"posx":30.892,"posy":11.625,"posz":16.340,"posa":0.000,"feed":500.00,"vel":500.00,"unit":1,"coor":1,"dist":0,"frmo":1,"stat":5}}
	{"sr":{"posx":30.274,"posy":11.625,"posz":15.849,"posa":0.000,"feed":500.00,"vel":500.00,"unit":1,"coor":1,"dist":0,"frmo":1,"stat":5}}
	{"qr":4,"qi":3,"qo":3}
	{"sr":{"posx":29.551,"posy":11.625,"posz":15.442,"posa":0.000,"feed":500.00,"vel":500.00,"unit":1,"coor":1,"dist":0,"frmo":1,"stat":5}}
	{"sr":{"posx":28.766,"posy":11.625,"posz":15.173,"posa":0.000,"feed":500.00,"vel":500.00,"unit":1,"coor":1,"dist":0,"frmo":1,"stat":5}}
	{"qr":4,"qi":3,"qo":3}
	{"sr":{"posx":27.989,"posy":11.625,"posz":15.033,"posa":0.000,"feed":500.00,"vel":500.00,"unit":1,"coor":1,"dist":0,"frmo":1,"stat":5}}
	{"sr":{"posx":27.160,"posy":11.625,"posz":15.014,"posa":0.000,"feed":500.00,"vel":500.00,"unit":1,"coor":1,"dist":0,"frmo":1,"stat":5}}
	{"qr":4,"qi":3,"qo":3}
	{"sr":{"posx":26.341,"posy":11.625,"posz":15.144,"posa":0.000,"feed":500.00,"vel":500.00,"unit":1,"coor":1,"dist":0,"frmo":1,"stat":5}}
	{"sr":{"posx":25.590,"posy":11.625,"posz":15.390,"posa":0.000,"feed":500.00,"vel":500.00,"unit":1,"coor":1,"dist":0,"frmo":1,"stat":5}}
	{"sr":{"posx":24.852,"posy":11.625,"posz":15.766,"posa":0.000,"feed":500.00,"vel":500.00,"unit":1,"coor":1,"dist":0,"frmo":1,"stat":5}}
	{"qr":4,"qi":3,"qo":3}
	{"sr":{"posx":24.186,"posy":11.625,"posz":16.261,"posa":0.000,"feed":500.00,"vel":500.00,"unit":1,"coor":1,"dist":0,"frmo":1,"stat":5}}
	{"sr":{"posx":23.642,"posy":11.625,"posz":16.834,"posa":0.000,"feed":500.00,"vel":500.00,"unit":1,"coor":1,"dist":0,"frmo":1,"stat":5}}
	{"qr":4,"qi":3,"qo":3}
	{"sr":{"posx":23.171,"posy":11.625,"posz":17.516,"posa":0.000,"feed":500.00,"vel":500.00,"unit":1,"coor":1,"dist":0,"frmo":1,"stat":5}}
	{"sr":{"posx":22.811,"posy":11.625,"posz":18.263,"posa":0.000,"feed":500.00,"vel":500.00,"unit":1,"coor":1,"dist":0,"frmo":1,"stat":5}}
	{"qr":4,"qi":3,"qo":3}
	{"sr":{"posx":22.603,"posy":11.625,"posz":19.026,"posa":0.000,"feed":500.00,"vel":500.00,"unit":1,"coor":1,"dist":0,"frmo":1,"stat":5}}
[... 20 lines cut ...]
	{"sr":{"posx":30.666,"posy":11.625,"posz":23.858,"posa":0.000,"feed":500.00,"vel":500.00,"unit":1,"coor":1,"dist":0,"frmo":1,"stat":5}}
	{"sr":{"posx":31.239,"posy":11.625,"posz":23.314,"posa":0.000,"feed":500.00,"vel":500.00,"unit":1,"coor":1,"dist":0,"frmo":1,"stat":5}}
	{"qr":14,"qi":0,"qo":3}
	{"sr":{"posx":31.734,"posy":11.625,"posz":22.648,"posa":0.000,"feed":500.00,"vel":500.00,"unit":1,"coor":1,"dist":0,"frmo":1,"stat":5}}
	{"sr":{"posx":32.110,"posy":11.625,"posz":21.910,"posa":0.000,"feed":500.00,"vel":500.00,"unit":1,"coor":1,"dist":0,"frmo":1,"stat":5}}
	{"qr":17,"qi":0,"qo":3}
	{"sr":{"posx":32.356,"posy":11.625,"posz":21.159,"posa":0.000,"feed":500.00,"vel":500.00,"unit":1,"coor":1,"dist":0,"frmo":1,"stat":5}}
	{"sr":{"posx":32.486,"posy":11.625,"posz":20.340,"posa":0.000,"feed":500.00,"vel":500.00,"unit":1,"coor":1,"dist":0,"frmo":1,"stat":5}}
	{"qr":20,"qi":0,"qo":3}
	{"sr":{"posx":32.467,"posy":11.625,"posz":19.511,"posa":0.000,"feed":500.00,"vel":500.00,"unit":1,"coor":1,"dist":0,"frmo":1,"stat":5}}
	{"sr":{"posx":32.327,"posy":11.625,"posz":18.734,"posa":0.000,"feed":500.00,"vel":500.00,"unit":1,"coor":1,"dist":0,"frmo":1,"stat":5}}
	{"sr":{"posx":32.058,"posy":11.625,"posz":17.949,"posa":0.000,"feed":500.00,"vel":500.00,"unit":1,"coor":1,"dist":0,"frmo":1,"stat":5}}
	{"qr":23,"qi":0,"qo":3}
	{"sr":{"posx":31.651,"posy":11.625,"posz":17.226,"posa":0.000,"feed":500.00,"vel":500.00,"unit":1,"coor":1,"dist":0,"frmo":1,"stat":5}}
	{"sr":{"posx":31.160,"posy":11.625,"posz":16.608,"posa":0.000,"feed":500.00,"vel":500.00,"unit":1,"coor":1,"dist":0,"frmo":1,"stat":5}}
	{"qr":26,"qi":0,"qo":3}
	{"sr":{"posx":30.550,"posy":11.625,"posz":16.046,"posa":0.000,"feed":500.00,"vel":500.00,"unit":1,"coor":1,"dist":0,"frmo":1,"stat":5}}
	{"sr":{"posx":29.891,"posy":11.625,"posz":15.594,"posa":0.000,"feed":500.00,"vel":500.00,"unit":1,"coor":1,"dist":0,"frmo":1,"stat":5}}
	{"qr":29,"qi":0,"qo":3}
	{"sr":{"posx":29.129,"posy":11.625,"posz":15.283,"posa":0.000,"feed":500.00,"vel":500.00,"unit":1,"coor":1,"dist":0,"frmo":1,"stat":5}}
	{"sr":{"posx":28.326,"posy":11.625,"posz":15.077,"posa":0.000,"feed":500.00,"vel":500.00,"unit":1,"coor":1,"dist":0,"frmo":1,"stat":5}}
	{"sr":{"posx":27.549,"posy":11.625,"posz":15.003,"posa":0.000,"feed":500.00,"vel":410.00,"unit":1,"coor":1,"dist":0,"frmo":1,"stat":5}}
	{"sr":{"posx":27.500,"posy":11.625,"posz":15.000,"posa":0.000,"feed":500.00,"vel":0.00,"unit":1,"coor":1,"dist":0,"frmo":1,"stat":3}}
	{"qr":32,"qi":0,"qo":3}
```

cncgoko, 2016-03-30:
Any update on this issue ?

ril3y, 2026-11-16:
Fixed in 450.01: G2/G3 in G18 now follow the standard direction (clockwise seen from +Y), so G3 runs the requested short arc. G18 programs written for the old TinyG direction need G2 and G3 swapped.

Fix: https://github.com/synthetos/TinyG/commit/98bc17a2 (first released in [450.01](https://github.com/synthetos/TinyG/releases/tag/450.01))
Regression test: `tests/hil/tests/bugs/test_e029_g18_arc_direction.py`

closed 2026-11-16 by ril3y