synthetos/TinyG #143: Communication loss between TinyG and Coolterm

troisiemetype, 2015-10-17:
I'm experiencing a strange thing with TinyG and coolterm: while running a program (Gcode file), the connection is sudenly lost. At first it happenned at an end of line, after a gcode instruction (that is: I suppose so beacause it happens at the end of a move), but once it hapened right in the middle of a straight line (move G1 from Y105 to Y-5, stopping at about Y40).
Closing and reopening connection make the move start back, as if it was just paused, sometimes with a move in G0 instead of G1. Sometimes this G0 instead of G1 move happens immediately, sometimes after several moves, sometimes not.
At first I tought it was a problem with the flow control, as the setting was $ex==1 and XON marked on Coolterm. But having set back $ex to 2 and CTS to on, the problem persists.
I had the problem several times, and then had another one that I think is linked, even if I don't understand how: doing an homing (G28.2X0Y0Z0) X- switch has not been detecting, and motor forcing. I tried to fire the other switch, but without any result. But once the tinyG shut down and ligted again, the homing happend whitout problem.

here are the settings of my card:
```
$$
[fb]  firmware build            440.14
[fv]  firmware version            0.97
[hp]  hardware platform           1.00
[hv]  hardware version            8.00
[id]  TinyG ID                    2X2660-FVT
[ja]  junction acceleration 2000000 mm
[ct]  chordal tolerance           0.0010 mm
[sl]  soft limit enable           0
[st]  switch type                 1 [0=NO,1=NC]
[mt]  motor idle timeout         60.00 Sec
[ej]  enable json mode            0 [0=text,1=JSON]
[jv]  json verbosity              2 [0=silent,1=footer,2=messages,3=configs,4=linenum,5=verbose]
[js]  json serialize style        1 [0=relaxed,1=strict]
[tv]  text verbosity              1 [0=silent,1=verbose]
[qv]  queue report verbosity      0 [0=off,1=single,2=triple]
[sv]  status report verbosity     1 [0=off,1=filtered,2=verbose]
[si]  status interval           250 ms
[ec]  expand LF to CRLF on TX     0 [0=off,1=on]
[ee]  enable echo                 0 [0=off,1=on]
[ex]  enable flow control         2 [0=off,1=XON/XOFF, 2=RTS/CTS]
[baud] USB baud rate              5 [1=9600,2=19200,3=38400,4=57600,5=115200,6=230400]
[net] network mode                0 [0=master]
[gpl] default gcode plane         0 [0=G17,1=G18,2=G19]
[gun] default gcode units mode    1 [0=G20,1=G21]
[gco] default gcode coord system  1 [1-6 (G54-G59)]
[gpa] default gcode path control  2 [0=G61,1=G61.1,2=G64]
[gdi] default gcode distance mode 0 [0=G90,1=G91]
[1ma] m1 map to axis              0 [0=X,1=Y,2=Z...]
[1sa] m1 step angle               1.800 deg
[1tr] m1 travel per revolution   60.0000 mm
[1mi] m1 microsteps               8 [1,2,4,8]
[1po] m1 polarity                 0 [0=normal,1=reverse]
[1pm] m1 power management         2 [0=disabled,1=always on,2=in cycle,3=when moving]
[2ma] m2 map to axis              0 [0=X,1=Y,2=Z...]
[2sa] m2 step angle               1.800 deg
[2tr] m2 travel per revolution   60.0000 mm
[2mi] m2 microsteps               8 [1,2,4,8]
[2po] m2 polarity                 1 [0=normal,1=reverse]
[2pm] m2 power management         2 [0=disabled,1=always on,2=in cycle,3=when moving]
[3ma] m3 map to axis              1 [0=X,1=Y,2=Z...]
[3sa] m3 step angle               1.800 deg
[3tr] m3 travel per revolution   60.0000 mm
[3mi] m3 microsteps               8 [1,2,4,8]
[3po] m3 polarity                 0 [0=normal,1=reverse]
[3pm] m3 power management         2 [0=disabled,1=always on,2=in cycle,3=when moving]
[4ma] m4 map to axis              2 [0=X,1=Y,2=Z...]
[4sa] m4 step angle               1.800 deg
[4tr] m4 travel per revolution    8.0000 mm
[4mi] m4 microsteps               8 [1,2,4,8]
[4po] m4 polarity                 1 [0=normal,1=reverse]
[4pm] m4 power management         2 [0=disabled,1=always on,2=in cycle,3=when moving]
[xam] x axis mode                 1 [standard]
[xvm] x velocity maximum       6000 mm/min
[xfr] x feedrate maximum       2400 mm/min
[xtn] x travel minimum            0.000 mm
[xtm] x travel maximum          510.000 mm
[xjm] x jerk maximum            500 mm/min^3 * 1 million
[xjh] x jerk homing            1000 mm/min^3 * 1 million
[xjd] x junction deviation        0.0500 mm (larger is faster)
[xsn] x switch min                3 [0=off,1=homing,2=limit,3=limit+homing]
[xsx] x switch max                2 [0=off,1=homing,2=limit,3=limit+homing]
[xsv] x search velocity        1000 mm/min
[... 103 lines cut ...]
[g59y] g59 y offset               0.000 mm
[g59z] g59 z offset               0.000 mm
[g59a] g59 a offset               0.000 deg
[g59b] g59 b offset               0.000 deg
[g59c] g59 c offset               0.000 deg
[g92x] g92 x offset               0.000 mm
[g92y] g92 y offset               0.000 mm
[g92z] g92 z offset               0.000 mm
[g92a] g92 a offset               0.000 deg
[g92b] g92 b offset               0.000 deg
[g92c] g92 c offset               0.000 deg
[g28x] g28 x position             0.000 mm
[g28y] g28 y position             0.000 mm
[g28z] g28 z position             0.000 mm
[g28a] g28 a position             0.000 deg
[g28b] g28 b position             0.000 deg
[g28c] g28 c position             0.000 deg
[g30x] g30 x position             0.000 mm
[g30y] g30 y position             0.000 mm
[g30z] g30 z position             0.000 mm
[g30a] g30 a position             0.000 deg
[g30b] g30 b position             0.000 deg
[g30c] g30 c position             0.000 deg
tinyg [mm] ok> 
```

stg, 2015-10-17:
I've had this issue a lot and there were two distinct reasons.

1) According to the author, serial-port-json-server has this issue on Windows with the latest version. If memory serves me, 1.80 is recommended for Windows for now. A new version seems to be drawing near.

The next one only applies to subtractive cnc machines such as mills, routers, lathes or anything that otherwise involves plenty of rubbing between different materials.

2) ESD. Seriously. A CNC mill can build up some pretty mind boggling levels static electricity. When cutting PU-foam (model board) I got zapped several times with enough juice to scare the living daylight out of me. Getting zapped by my 25kV/25mA laser cutter power supply was nasty, but not that scary, which should tell you something about just how bad it can get. Static electricity can creep along wires and discharge at the controller. While the discharge, due to high surface impedance of the plastic cable housing, is typically not strong enough to fry the electronics it is more than enough to screw up the processor in all the strangest ways. I've had memory changes, g-code misinterpretations, wonky motion, crashes, etc. but most frequently by far - communication losses!

By grounding the spindle and the control board ghetto style with crocodile clip lab wires my problems nearly went away completely. The proper solution of course would be to encase the controller in a grounded metal box with ESD cable inserts (slightly conductive plastic) which will catch any ESD discharges before they can reach the internal electronics.

It's easy enough to try if you have ground available at your power outlets, so perhaps give it a shot?

johnlauer, 2015-10-17:
The new version 1.86 of SPJS is out so please use that. BTW, there has never been a disconnect problem on SPJS. The only problem on Windows was the serial library in 1.83 having packet loss. That library is no longer used.

stg, 2015-10-17:
Then forget completely about my first point, this must have been a misunderstanding between me and John on the SPJS issues forum.

Nice to hear 1.86 is out, very much looking forward to trying that, as feedhold and flush queue started working for me in version 1.82 and hasn't before. It's a pretty useful function for sure ;)

troisiemetype, 2015-10-19:
Hello,

As I use a very old macbook pro, I think the first point isn't o f concern for me. More, it wouldn't explain why I've never had this problem before last week, given that I've not changed anything in my set up for a few month. It really come out of nowhere!

The second point make more sense to me: for the first time since I use my router, i've seen sparkles made by the smindle motor brushes. I've been wondering about static electricity. I'm gonna try to ground the aluminium structure and the spindle. But where to connect it? Do you connect it to ground plug in the wall? The power supply for the TinyG and the steppers is grounded, but the spindle is not. And where to connect ths ground on the TinyG board itself?

stg, 2015-10-19:
Even if you think it is grounded, sometimes it is not. For example, my laptop has an isolating transformer that does not provide actual ground - so no ground via USB. The power supply for my G2 controller does not even have ground pins, only line and neutral. While neutral is at ground, in Sweden there is no keying for polarity meaning devices are completely ignorant as to which pin is ground unless they have the third protective earth connection.

Anyways, what I did was run a wire from ground on the controller to two crocodile clips attaching one to ground pins in the nearest outlet and the other to the spindle motor assembly. I still get the problem when manually vacuuming and standing too close to the controller, as chips are heavily charged, transfer to the vacuum, to me and to the controller. As this is reproducible, I see it as the final confirmation of having had ESD issues. Again, a proper Au or conductive plastic casing for the device is the proper method. I'll be doing just that once I feel I can fully rely on the loaded firmware.

aldenhart, 2015-12-06:
Is it OK to close this issue?

troisiemetype, 2016-02-08:
I think it can be: I didn't have the problem again.
Thank you.

ril3y, 2021-04-09:
Closed due to inactivity

closed 2021-04-09 by ril3y