synthetos/TinyG #144: Gcode does not always load correctly

matthijskooijman, 2015-10-28:
While using the grbl workspace this afternoon, I found that a new gcode file is not always loaded correctly. I initially observed this while milling, when loading a new file after finishing another. Later, I reproduced this even when just switching between files in a clean grbl workspace, without any serial port attached or any milling happening (this did require a switching between files a few times, it usually worked). When the file was not loaded correctly, I think the 3D viewer did change (I'm not sure if it actually showed things correctly, though), but the contents of the Gcode panel did not change. I also think that the "File loaded" dialog was not displayed in this case, but I'm not entirely sure.

I tried reproducing this in the tinyg workspace just now, without any luck. However, I couldn't reproduce this in the grbl workspace again just now, so this might end up being a tricky problem to debug...

ril3y, 2015-10-28:
This is an chilipeppr issue not an TinyG issue.  I would suggest you check out the chilipeppr google+ community as its where you can get support and report issues on Chilipeppr.

closed 2015-10-28 by ril3y

matthijskooijman, 2015-10-28:
W00ps, wrong bugtracker. I rememberd that chilipeppr.com/grbl bugs should be reported under tinyg, but that was https://github.com/chilipeppr/tinyg

Sorry for the noise and the quick response :-)

ril3y, 2015-10-29:
No problem at all. It all starts to run together :)