synthetos/TinyG #146: Zero backoff distance when homing appears to be wrong

rickcaddell, 2015-11-22:
This is firmware I am using.
[fb]  firmware build            440.20
[fv]  firmware version            0.97
[hp]  hardware platform           1.00
[hv]  hardware version            8.00
[id]  TinyG ID                    3X3566-JUU
...
tinyg [inch] 

The Zero back off value is 0.1 inch. (set using JSON command {xzb:0.1})
The units mode is set to inch.

When I initiate a home sequence with a zero back off value of 0.1 inch , the machine actually backs off 2.5 inches or so. If I set the zero backoff to 0.004 (0.1 / 25.4) it backs off about 0.1 inch. 

It appears that there is some mixup with metric units.

aldenhart, 2015-12-06:
Thanks for the tip. Looking into it.

rickcaddell, 2015-12-06:
Something odd is going on. I have two TinyG's. One in my lab that is connected to a couple of motors and switches that I use for testing, and one on a machine in my shop (which I just got running). Both machines have the same firmware build, firmware version, hardware platform and hardware version.
The lab machine does not do the back off correct, but the shop machine does.Not sure what is going on here.

ril3y, 2015-12-06:
rickcaddell,

The configs are the same?  We really would need some more information on
what is going on.  I would say that (having done this exact thing) is one
tinyg is wired correctly and one is not.  Perhaps one of the switches is
bad?

rickcaddell, 2015-12-06:
The configs are definitely the same, and I checked the switch. The TinyG that does not seem to work is an older one, and

I did have problems flashing it to the latest version, maybe it did not entirely reprogram, or there is something wrong with

the hardware.

 

I would not spend any more time on this right now, one board works so your software is probably OK.

 

I will re-flash and retest, and if need be try the old board on my shop machine  and  let you know if it still does not work. 

 

Rick

closed 2016-02-03 by rickcaddell