synthetos/TinyG #153: Queue Reports and Staus Reports not correct in different situations. Even in 442.04

rickcaddell, 2016-02-07:
I have been fighting a problem for quite some time and have tracked it down to the queue report and
status reports not working the same in all situations.

I like to step through my GCode programs, meaning when I hit a step button it sends one command to the TinyG. When running continuously I send as many as possible, keeping in mind the buffer constriants.

The problem is I cannot figrue out from the queue reports or status reports when a GCode block has actually finished executing. In step mode the queue reports appear to be wrong, skipping any block that does not cause motion but the status reports look right. When running contiuoualy, the queue reports look right but the status report for any block that does not cause motion is missing.

I recently downloaded the 442.04 version because it has the packet mode, thinking that this would help. I really like the new firmware updater by the way. The packet mode did not act any different than the old mode.

On reset I get the following message indicating I have the 442.04 firmware :

{"r":{"fv":0.970,"fb":442.04,"hp":1,"hv":8,"id":"3X3566-KBR","msg":"SYSTEM READY"},"f":[1,0,0,225]}

When I turn on packet mode, I notice that the footer is still like the old footer. That seems to contradict the documentation which describes only three parameters.

{rxm:1}
{"r":{"rxm":1},"f":[1,100,8,4905]}

{rxm:n}
{"r":{"rxm":null},"f":[1,100,8,5855]}

This is a simple G Code program that works differently when stepping than running.

N0 G1 X.001 Y.001 Z0 F60
N1 M03
N2 G1 X0 Y0 Z0 F60

If I step through the program, the queue report does not make sense. It only shows 2 blocks being queued and 2 being dequeued. I think it should be three. The ones for block N1 (a non motion block) are missing. The status reports look correct showing all three lines.

When Stepping :

{"gc":"N0 G1 X0.001 Y0 Z0 F60"}
{"r":{"gc":"N0G1X0.001Y0Z0F60"},"f":[1,0,32,2084]}
{"sr":{"line":0,"posx":0.000,"posy":0.000,"posz":0.000,"mpox":0.000,"mpoy":0.000,"mpoz":0.000,"vel":0.00,"stat":5,"coor":1,"unit":0}}
{"qr":31,"qi":1,"qo":0}
{"sr":{"line":0,"posx":0.001,"posy":0.000,"posz":0.000,"mpox":0.015,"mpoy":0.000,"mpoz":0.000,"vel":0.63,"stat":5,"coor":1,"unit":0}}
{"sr":{"line":0,"posx":0.001,"posy":0.000,"posz":0.000,"mpox":0.025,"mpoy":0.000,"mpoz":0.000,"vel":0.00,"stat":3,"coor":1,"unit":0}}
{"qr":32,"qi":0,"qo":1}
{"gc":"N1 M3"}
{"r":{"gc":"N1M3","n":1},"f":[1,0,15,3441]}
{"sr":{"line":1,"posx":0.001,"posy":0.000,"posz":0.000,"mpox":0.025,"mpoy":0.000,"mpoz":0.000,"vel":0.00,"stat":3,"coor":1,"unit":0}}
{"gc":"N2 G1 X0 Y0 Z0 F60"}
{"r":{"gc":"N2G1X0Y0Z0F60","n":2},"f":[1,0,28,8966]}
{"sr":{"line":2,"posx":0.001,"posy":0.000,"posz":0.000,"mpox":0.025,"mpoy":0.000,"mpoz":0.000,"vel":0.00,"stat":5,"coor":1,"unit":0}}
{"qr":31,"qi":1,"qo":0}
{"sr":{"line":2,"posx":0.000,"posy":0.000,"posz":0.000,"mpox":0.009,"mpoy":0.000,"mpoz":0.000,"vel":0.62,"stat":5,"coor":1,"unit":0}}
{"sr":{"line":2,"posx":0.000,"posy":0.000,"posz":0.000,"mpox":0.000,"mpoy":0.000,"mpoz":0.000,"vel":0.00,"stat":3,"coor":1,"unit":0}}
{"qr":32,"qi":0,"qo":1}

When Running the queue report looks correct but the status report for block N1 is missing.
{"gc":"N0 G1 X0.001 Y0 Z0 F60"}
{"gc":"N1 M3"}
{"gc":"N2 G1 X0 Y0 Z0 F60"}
{"r":{"gc":"N0G1X0.001Y0Z0F60"},"f":[1,0,32,2084]}
{"sr":{"line":0,"posx":0.000,"posy":0.000,"posz":0.000,"mpox":0.000,"mpoy":0.000,"mpoz":0.000,"vel":0.00,"stat":5,"coor":1,"unit":0}}
{"qr":31,"qi":1,"qo":0}
{"r":{"gc":"N1M3","n":1},"f":[1,0,15,3441]}
{"sr":{"line":0,"posx":0.000,"posy":0.000,"posz":0.000,"mpox":0.000,"mpoy":0.000,"mpoz":0.000,"vel":0.08,"stat":5,"coor":1,"unit":0}}
{"qr":30,"qi":1,"qo":0}
{"r":{"gc":"N2G1X0Y0Z0F60","n":2},"f":[1,0,28,8966]}
{"sr":{"line":0,"posx":0.000,"posy":0.000,"posz":0.000,"mpox":0.002,"mpoy":0.000,"mpoz":0.000,"vel":0.29,"stat":5,"coor":1,"unit":0}}
{"qr":29,"qi":1,"qo":0}
{"sr":{"line":0,"posx":0.001,"posy":0.000,"posz":0.000,"mpox":0.023,"mpoy":0.000,"mpoz":0.000,"vel":0.34,"stat":5,"coor":1,"unit":0}}
{"qr":30,"qi":0,"qo":1}
{"qr":31,"qi":0,"qo":1}
{"sr":{"line":2,"posx":0.001,"posy":0.000,"posz":0.000,"mpox":0.023,"mpoy":0.000,"mpoz":0.000,"vel":0.29,"stat":5,"coor":1,"unit":0}}
{"sr":{"line":2,"posx":0.000,"posy":0.000,"posz":0.000,"mpox":0.001,"mpoy":0.000,"mpoz":0.000,"vel":0.29,"stat":5,"coor":1,"unit":0}}
{"sr":{"line":2,"posx":0.000,"posy":0.000,"posz":0.000,"mpox":0.000,"mpoy":0.000,"mpoz":0.000,"vel":0.00,"stat":3,"coor":1,"unit":0}}
{"qr":32,"qi":0,"qo":1}

My questions are :

Is the packet mode in 442.04 as stated in the documentation? (It says it is in 441.xx.)
Should the queue report show all blocks going into and out of the queue?
Should status reports show all blocks being executed?

I am at the end of a rope. I need to know when a block has executed because I am updating a table of G code blocks to show the user where the machine is in the program and updating a drawing. Right now I am stuck. Any ideas?

johnlauer, 2016-02-07:
You should just do it the way I do it in Serial Port JSON Server's buffer flow mechanism (or consider using SPJS) where I simply look for the r:{} response for each line to decrement which lines got processed. It's the only reliable method I found. I initially tried processing qr reports and it didn't work well.

rickcaddell, 2016-02-08:
John ,

 

Thanks for the reply.

Below is a screen shot of my application. The table on the left is the program being run with the drawing to the right.

I am trying to update the table and drawing as the program runs. Blue in the drawing is uncut, Green is cut, brown

is the current block and yellow is how much of that block has been cut.

 

Note the first column in the table. This is the status of the block. If the block is in the TinyG queue, it is marked with a '*'.

If it is the current block being executed it is marked with '>' and if it has already been executed it is marked with '|'.

 

The r:{} works great for marking the records that are in the TinyG Queue, but does not indicate when a block is

started or finished. I was hoping to use the queue report, but that does not seem reliable. What I am trying now

is using the status report with the line parameter  eg.  {"sr":{"line":1,". Since this report is on a time basis some

lines may not be reported, like blocks that do not have movement (M3) or when contouring very small blocks. 

I think what I am going to do is monitor the line numbers in the status report and if I see some being skipped

I will just mark them as finished.

 

 

Thanks again for the reply. It is good to talk with other people using the TinyG.

 

My application works well for me but when I make changes I use the Visual Studio compiler.

I am planning to make it so users can add/delete/edit controls like buttons, and have the ability to run 

macros on events like TinyG messages and button clicks, so a compiler will not needed. When finished

I plan to give it to  Synthetos so others can use it.

 

Rick (Old Man)

closed 2016-03-08 by rickcaddell