synthetos/TinyG #154: Smoke test fail

tomfitzphilly, 2016-02-15:
Sorry - I searched and couldn't fine anything.  Ran $test=1 and got this (although a lot longer):
qr:15
{"er":{"fb":440.20,"st":31,"msg":"Move time is NAN"}}
{"er":{"fb":440.20,"st":31,"msg":"Move time is NAN"}}
{"er":{"fb":440.20,"st":31,"msg":"Move time is NAN"}}
{"er":{"fb":440.20,"st":28,"msg":"Failed to get planner buffer"}}
{"er":{"fb":440.20,"st":28,"msg":"Failed to get planner buffer"}}
{"er":{"fb":440.20,"st":28,"msg":"Failed to get planner buffer"}}

And then two motor controls stopped working.

My guess is a fried chip on each channel, I just can't figure out why.

Any thoughts?

ril3y, 2016-02-16:
I am not sure why.  I am also not sure its fried.  Try $defa=1 and see if
we can reset to defaults.

tomfitzphilly, 2016-02-17:
That worked.  Damned if I understand what happened.

Thanks
Tom

ril3y, 2016-02-17:
Its not common but what happened was the EEPROM that contained your
configuration values became corrupted.  $defa=1 will reset your EEPROM
settings values back to their "factory" settings.

I am not sure how this became corrupt but glad its working!

Riley

tomfitzphilly, 2016-02-17:
Thanks again. I've ordered 2 more of these boards. I really like them. One
more for a small jeweler's mill I have and one for a small laser I am
building. This board tied to chili pepper makes my life so much easier.

ril3y, 2021-04-09:
Closed due to inactivity

closed 2021-04-09 by ril3y