synthetos/TinyG #159: Filtered Status Reports

rickcaddell, 2016-02-27:
Firmware Version : 0.970
Firmware Build : 446.19
Hardware Platform : 1
Hardware Version : 8
ID : 3X3566-KBR

I ran into something on the latest Edge version. The satus reports seem to be stuck in filtered mode.
I set the mode to verbose, then reset the TinyG, then query sv.
{sv:n}
 {r:{sv:2},f:[3,0,24]}
It says it is in verbose, but the status reports are filtered.
 {sr:{posx:1.386,vel:4.07,mpox:35.215}}

if I request a status report :
{sr:n}
I get : sr:{sr:line:0,posx:0.000,posy:0.000,posz:0.000,posa:0.000,feed:0.00,vel:0.00,unit:0,coor:1,dist:0,admo:0,frmo:1,momo:4,stat:3,mpox:0.000,mpoy:0.000,mpoz:0.000}},f:[3,0,24]}

This doesn't look correct either.

I set the status report format to :
{sr:{line:t,posx:t,posy:t,posz:t,mpox:t,mpoy:t,mpoz:t,vel:t,stat:t,coor:t}
{r:{sr:line:true,posx:true,posy:true,posz:true,mpox:true,mpoy:true,mpoz:true,vel:true,stat:true,coor:true}},f:[3,0,24]}

The report format doesn't call for dist:0,admo:0,frmo:1,momo:4 but they are in the report.

Should I be setting these to false? I think I remember reading in past documentatin on status reports that there was a way clear the format. I did not see this in the new documentation. DId I miss it or is there still a way to do this?

Good job on reports by the way, I really like how they are setup.

Old Man

aldenhart, 2016-02-28:
Thanks for the heads up. Will take a look.

aldenhart, 2016-02-29:
OK. I had really changed that section of the code and made a dumb mistake. It's fixed in edge-0.97, build 448.03 (and later). Please let me know if you find anything else.

closed 2016-05-16 by rickcaddell