synthetos/TinyG #161: Queue report does not seem to work properly

rickcaddell, 2016-02-28:
I have always had trouble with my software keeping in sync with the TinyG as far as knowing
what GCode block is being started, and finished. The following illustrates my problem.

Run the following program :
N1 G0 X.01
N2 M03
N3 M05
N4 G0 X.03

These are the resulting queue reports:
{qr:31,qi:1,qo:0}
{qr:30,qi:1,qo:0}
{qr:29,qi:1,qo:0}
{qr:28,qi:1,qo:0}
{qr:29,qi:0,qo:1}
{qr:31,qi:0,qo:2}
{qr:32,qi:0,qo:1}

Looks good to me, 4 in 4 out

Run the following program : Same blocks in a different order.
N1 M03
N2 G0 X.01
N3 G0 X.03
N4 M05

These are the resulting queue reports:
{qr:31,qi:1,qo:0}
{qr:30,qi:1,qo:0}
{qr:29,qi:1,qo:0}
{qr:30,qi:0,qo:1}
{qr:31,qi:0,qo:1}
{qr:32,qi:0,qo:1}

Same blocks in different order but only 3 in and 3 out

Run the following program :
N1 M03
N2 M05
N3 G0 X.01
N4 G0 X.03

{qr:31,qi:1,qo:0}
{qr:30,qi:1,qo:0}
{qr:31,qi:0,qo:1}
{qr:32,qi:0,qo:1}

Same blocks in different order but only 2 in and 2 out

I would like to use queue reports to keep my program in sync with the TinyG but because of the above I can't.

A better way would be to use line numbers. The first parameter would be the same (available slots) the second
parameter would be the last line number going into the queue and the third would be the line number of the last 
line finised.

Running one command at a time might look like:
N1 M03
{qr:31,li:1,lo:1}
N2 M05
{qr:31,li:2,lo:2}
N3 G0 X.01
{qr:31,li:3,lo:3}
N4 G0 X.03
{qr:31,li:4,lo:4}

Streaming might look like"
N1 M03
{qr:31,li:1,lo:0}
N2 M05
{qr:31,li:2,lo:0}
N3 G0 X.01
{qr:31,li:3,lo:0}
N4 G0 X.03
{qr:31,li:4,lo:0}
{qr:31,li:0,lo:1}
{qr:31,li:0,lo:3} <--- Block 2 was not shown because it executed so fast, but that is OK
{qr:31,li:0,lo:4}

Note that block was not show in the queue report, but that doesn't matter because I know the kast
block is 3, so I can take care of 2. This would happen so quickly an operator would not notice anyway.

This would be the ultimate queue report for me. It would solve my syncing problems.

I can't say enough about how much trouble I have had trying to keep in sync with the TinyG. I am
probobly doing things others don't like allow stepping through a program, which is one of the things
that is hard to sync, but being able to step through a program to prove it out, or if there are
problems, is invaluable.

Old Man

aldenhart, 2016-02-29:
Looking into this.

aldenhart, 2016-02-29:
OK. I had really changed that section of the code and made a dumb mistake. It's fixed in edge-0.97, build 448.03 (and later). Please let me know if you find anything else.

rickcaddell, 2016-02-29:
When will build 448.03 be availible on the web?
I would love to test ths.

aldenhart, 2016-02-29:
My earlier entry in this issue was intended for Issue #159 - Filtered Status Reports. 

No changes to queue reports have occurred in build 448.03. Please See Issue #159

closed 2016-05-02 by rickcaddell