synthetos/TinyG #164: When using external drivers, how do I change the width of the pulses?

tglaria, 2016-03-08:
I'm guessing that the pulses for external drivers are made by the DDA timer interrupt (For AVR,  `ISR(TIMER_DDA_ISR_vect)` in stepper.c file).
On each port/motor, a pin is driven high  and then driven low at each DDA timer interruption as long as the number of pulses is less than what's configured.
In the comments I see:
`// ~ 5 uSec pulse width
// ~ 4 uSec
// ~ 3 uSec
// ~ 2 uSec`

Now, how are those values specified?
How could I change them?

I wasn't able to find where could I modify the width of the pulses.

Are those fixed?

DanalEstes, 2016-03-09:
Not questioning the question... you may need to change pulse widths... just a data point:

I am using TinyG with Geko G201X drivers (at 70V on the motor side).  I was very concerned about burning up something on one end or the other.   In the end, everything "just worked", with no changes on either side (beyond soldering screw terminal headers on the TinyG).  That also probably means that any Geko is going to work.

DanalEstes, 2016-03-09:
Also, if it helps, I happened to have a TinyG V8 hardware, build 97, and a scope handy.  Build 97 outputs right at 2.5us pulses.

tglaria, 2016-03-09:
I beleive I do need to change the width of the pulses, the thing I don't get is what are those comments (I don't see how there is a change of pulse width from one line of code to the other).

I tried putting some `nop` inside that ISR, but no luck.

DanalEstes, 2016-03-09:
Yeah, those comments have no real relation to the code, or the us of deleay, as far as I can tell.  

Disclosure: I am not a TinyG core contributor, just a user.  On the other hand, I have written multiple ISRs on the Atmel platform, including some real-time motion control.  

So take my credibility with a grain of salt.   

I am burningly curious, what is the need to change the width?

hackware, 2016-09-29:
DanalEstes, could you give me some information on your setup with the TinyG and Gecko drivers?
I am just starting out on cnc, and the TinyG looks good to me, but I am concerned about running NEMA 34 steppers straight from the TinyG. Your setup looks promising...
(30+ years electronics/systems, not scared of solder fumes :-)

william...

DanalEstes, 2016-09-30:
In the case of my Gecko's, everything "Just Worked" straight out of the
box.  I was concerned about 3.3V and 5V and so forth, and proceeded slowly
and cautiously.

Really, all I did was solder on the pin headers into the existing holes on
the TinyG, turn the onboard current pots all the way down, and wire the
direction and step lines to the Geckos as labeled, and the TinyG pins
labeled "Gnd" to Gecko "Common".

I am running the high side of the Geckos at 70V and have had absolutely no
issues.

Just as a reminder, these are Gecko G201X

ril3y, 2021-04-09:
Closed due to inactivity

closed 2021-04-09 by ril3y