synthetos/TinyG #166: Probe Anti-crash Feature

rickcaddell, 2016-03-19:
I use a spindle probe a lot. I have programs that probe to locate featues on a fixture for example. 

It is easy to crack one up. My suggestion is that when NOT in a probe cycle, (ie. when jogging, homing or executing a motion command like G0 or G1) if a probe touch is detected, do a feed hold and report a probe error with an error number indicating an unexpected probe touch.

This would keep accidental movement from damaging the probe.

aldenhart, 2016-03-20:
Good suggestion. I'm currently doing some work on probing (and a bunch of other things) and will see about including this capability. I will update as it becomes available for testing. It will appear in edge-0.97 when that's updated.

rickcaddell, 2016-03-20:
Alden,

 

Thanks for the consideration.

 

Old Man

ril3y, 2016-03-22:
Rick you should capture this feature request in an github issue.  Yes it
means you would have to get an github account :)

Riley

rickcaddell, 2016-03-22:
Riley,

 

I did post this on github as a TinyG Issue.

See issue #166

 

I still owe you for a TinyG, how can I get the money to you?

 

Rick

ril3y, 2021-04-09:
Closed due to inactivity

closed 2021-04-09 by ril3y