synthetos/TinyG #167: TinyG strange stepper movement

dgtlmoon, 2016-03-20:
Hi all

So I've been fiddling with this for a couple of days, I've got the setup tuned quite conservatively (I think)

The problem for me is that the stepper is running extremely erratically and none of the tuning seems to help. I tried to submit something to the TinyG Synthetos forum but when I log in the forum still says 'Login/register'.

Where to start? disabling micro-stepping seems to help a little bit, but overall not really making any progress

See the movement here https://www.youtube.com/watch?v=cMBSPbrQ-KM

Any suggestions?

```
[fv]  firmware version            0.97
[hp]  hardware platform           1.00
[fb]  firmware build            440.20
```

$x ( I dont find the command to show all configurations at https://github.com/synthetos/TinyG/wiki/TinyG-Configuration-for-Firmware-Version-0.97 does this page need updating?)

```
[xam] x axis mode                 1 [standard]
[xvm] x velocity maximum         30 in/min
[xfr] x feedrate maximum         30 in/min
[xtn] x travel minimum            0.000 in
[xtm] x travel maximum        10000.000 in
[xjm] x jerk maximum              5 in/min^3 * 1 million
[xjh] x jerk homing               2 in/min^3 * 1 million
[xjd] x junction deviation        0.0020 in (larger is faster)
[xsn] x switch min                0 [0=off,1=homing,2=limit,3=limit+homing]
[xsx] x switch max                0 [0=off,1=homing,2=limit,3=limit+homing]
[xsv] x search velocity          20 in/min
[xlv] x latch velocity            4 in/min
[xlb] x latch backoff             0.197 in
[xzb] x zero backoff              0.039 in
tinyg [inch] ok> 
```

DanalEstes, 2016-03-21:
You currently have your motor wired 
A1-Green A2-Red B1-Black B2-Blue
Try 
A1-Green A2-Black B1-Red B2-Blue

If that does not help, unwire all 4 motor wires, then turn the shaft by hand while pinching pairs of wires together.  One pair will make the motor MUCH more resistant to being turned.  Wire that pair to A1 A2.  Wire the other pair to B1 B2.

dgtlmoon, 2016-03-21:
Thanks for the reply , OK well spotted - I tried that, as you say green+black together made the motor harder to turn, so I've wired it as you say. (damn, I should have seen that!)

However now there is no movement when I issue a `{"gc":"G20 G91 G0  X10"}`

I see the green lights on the board light up but theres no movement, where to next?

troisiemetype, 2016-03-21:
It may seem evidence, but have you turn the pot so the motor have enough power running trough it?
If yes, Does another motor behave the same?

dgtlmoon, 2016-03-21:
Ok cool! Thanks, I managed to get some action but I'm finding it hard to configure, how do define "jerk maximum" ? if it's * million, that's a lot of jerk! I can't find any good references to what this value does in terms of hardware.

I had no problems with configuring GRBL setup, but this is a little trickier.

Most of the time the motor now just 'engages' I have had it working once but I lost the config and don't really understand what 'jerk' does

dgtlmoon, 2016-03-21:
Is it worth filing an issue about the jerk documentation?

closed 2016-03-21 by dgtlmoon