synthetos/TinyG #168: Broken Makefile on OS X

bdurbrow, 2016-03-22:
As of 3/22/2016, downloading and building the master branch on OS X fails.

The makefile in tinyg/default has a bunch of stuff like this in it, which make is choking on:

<<<<<<< HEAD:firmware/tinyg/default/Makefile
CFLAGS += -gdwarf-2 -std=gnu99 -Wall -DF_CPU=32000000UL -Os -funsigned-char -funsigned-bitfields -fpack-struct -fshort-enums
=======
CFLAGS += -gdwarf-2 -std=gnu99 -Wall -DF_CPU=32000000UL -Os -funsigned-char -funsigned-bitfields -fpack-struct -fshort-enums
>>>>>>> refs/heads/edge:firmware/tinyg/default/Makefile
CFLAGS += -MD -MP -MT $(*F).o -MF dep/$(@F).d

I don’t claim to be an expert on makefiles, or github in general, but this looks like a patch file didn’t get merged correctly???

I have edited the makefile to remove the offending lines; and I *think* I got it right - at least it now builds on my machine. I have not yet been able to test it due to an unrelated hardware issue (the board that I have has a 192a3u on it, and the headers for that from Atmel conflict with some of the #defines in TinyG - so I will have to do further editing of the source to get it to build for my hardware; but that's not a makefile related issue).

I have attached my version of the makefile to this issue report as a .txt file (github won't let me upload the file without a .txt extension - obviously, it doesn't have the .txt extension when it's in my build directory).

FWIW, I’m on a Mac, with OS X 10.11.3; Xcode 7.2.1 (7C1002); and CrossPack 20131216 installed.

[Makefile.txt](https://github.com/synthetos/TinyG/files/184163/Makefile.txt)

aldenhart, 2016-03-31:
The makefile used in the Atmel Studio 6 project is auto-generated by AS6. It's best to install that package and run from there. Instructions are on this wiki. I have not tried to build directly from the makefile, so I can't help you there.

closed 2016-03-31 by aldenhart

bdurbrow, 2016-04-05:
The package as distributed by Github contains an Xcode project for building on OS X. Either that should be removed if building on OS X won't be supported; or it should be maintained so that it works.

Obviously, I would prefer if OS X builds were continued to be supported. :-)

As I am on OS X, I can do that for you if you don't have a Mac to do the maintenance of the Xcode project and associated make file.