synthetos/TinyG #177: Firmware Updating Process

rickcaddell, 2016-05-16:
In various responses to issues I have brought up in the past, developers have said that they fixet the issue and was available in a certain version, but these never have shown up in the Firmware Update Downloads page. The following is the only versions I have seen on the page.

Branch 	Version 	Download Link
master 	440.20 	tinyg-master-440.20.hex
master-candidate 	440.19 	tinyg-master-440.19.hex
edge 	446.19 	tinyg-edge-446.19.hex

Am I looking in the right place for updates, or am I misunderstanding the release process.
How do I get the latest versions?

Old Man

madeinoz67, 2016-09-05:
wondering the same thing myself

ril3y, 2016-09-06:
This is the page to acquire firmware updates.
http://synthetos.github.io/

madeinoz67, 2016-09-06:
Yes and these are the same versions as outlined above.   What about latest stable firmware for production that fixes Arc G2/G3 issue?  From what I can see there has been no commits to stable Master branch for > 1Year From my testing the Arc issue appears to be fixed in Edge, but is not production ready for obvious reasons, also edge is broken on ChiliPeppr with it not reporting mPos

stg, 2016-09-07:
Temporary hotfix for mpos in status reports.

Problem: G2 does not support incremental status report configuration (everything needs to be on a single json line) resulting in a setup string that is too long for the configured buffer limits. The command gets stuck in serial-port-json-server buffers.

This is a hack, not a quality solution - but it will get Chilipeppr to play nice with edge by implementing rudimentary incremental status reports.

Edit report.cpp:

```c
stat_t sr_set_status_report(nvObj_t *nv) {
	uint8_t elements = 0;
	uint8_t count = 0;
	index_t status_report_list[NV_STATUS_REPORT_LEN];
	memcpy(status_report_list, sr.status_report_list, sizeof(status_report_list));
	for(count = 0; count < NV_STATUS_REPORT_LEN; count++) {
		if(status_report_list[count] == 0) break;
	}
	for (uint8_t i=0; i<NV_STATUS_REPORT_LEN; i++) {
		if (((nv = nv->nx) == NULL) || (nv->valuetype == TYPE_EMPTY)) break;
		if (nv->valuetype == TYPE_BOOL) {
			uint8_t found;
			for(found = 0; found < count; found++) {
				if(status_report_list[found] == nv->index) break;
			}
			
			if(fp_TRUE(nv->value)) {
					if(found == count) status_report_list[count++] = nv->index;
			} else {
					if(found < count) {
						memmove(&status_report_list[found], &status_report_list[found + 1], (--count - found) * sizeof(index_t));
						status_report_list[count] = 0;
					}
			}
			elements++;
		} else return (STAT_INPUT_VALUE_UNSUPPORTED);
	}
	memcpy(sr.status_report_list, status_report_list, sizeof(status_report_list));
	return(_populate_unfiltered_status_report());			// return current values
}
```

Make sure you use chilipeppr with "send no init cmds mode" (?v9=true) and add this to your serial port startup script:

```
{sr:{mpox:true,mpoy:true,mpoz:true}}
{qv:1}
{qr:n}
{sr:n}
```

aldenhart, 2016-09-07:
Thanks. We will be porting in the incremental status report behaviors from TinyG v8 edge into g2core. Please stay tuned.

closed 2017-06-30 by ril3y