synthetos/TinyG #178: Long moves do not stop on programmed value.

rickcaddell, 2016-05-16:
I am running : Firmware Version : 0.970, Firmware Build : 440.20, Hardware Version : 8

I have a large machine (X28, Y4=35), but only lately have been using bigger moves, and have found that long moves do not stop on programmed value.

To recreate this issue, Do G0 X24.452 it goes there correctly, then G0 X24.000 the machine actually goes to 24.001. The status report has posx:24.001 mpox:609.621. mpox should be 24.000 / 25.4 = 609.600. If the same command is given again G0 X24.000 then it goes to X24.000 and the status report is posx:24.000 mpox:609.600 which is correct.

If commanded back to the original position G0 X24.452 it actually goes to 2.451 instead of 2.452. The status report has posx:2.451 mpox:62.260. mpox should be mpox:62.281. If the same command is given again G0 X24.452 then it goes to G0 X24.452 and the status report is posx:2.452  pox:62.281 which is correct.

I believe that this error may be accumulative, in other words if many long moves are made the positions get off more and more. I have no good way of proving this, but I have noticed that after a lot of long moves X0 and or Y0 are not where they should be. 

I have dials on the ends of the lead screws that I can set to a marker after a home and squaring operation. Whenever I home or power up I can command the machine to X0Y0 and check the dials to see if anything has moved.

I can move the machine to X0Y0, set the dials, then run a program that has long moves, then move to X0Y0. The dials do not line up with the marks indicating that position has been lost. It may be something other than the issue described above, but it looks suspicious.

Old Man

solhuebner, 2020-01-27:
@rickcaddell do you know if this is still an issue?

ril3y, 2021-04-09:
Closed due to inactivity

closed 2021-04-09 by ril3y