synthetos/TinyG #179: Initial Power Up

harolddmorales, 2016-06-02:
Hi, 

I just purchased a TinyG for a side project at work. I have a 24v 6A power supply, a nema 17 stepper, and a small 12v fan. I connected the power as usual (+ to +, - to ground) and I used a Multi-meter to verify the volts and connections. 

When I plugged in the TinyG, the Blue light lit up and the red light started lighting up through SpOn, SpDir, then when it got to PWM it started smoking towards the reset button. It was hard to tell where the smoke came from, since there was no clear signs of damage. 

What did I do wrong and how to fix this issue. 

Thank you,
Harold M

Keith-W, 2016-10-22:
Did you ever figure out what happened?  I'm getting the feeling this forum is not that active since there are very few responses.  I have posted two questions / issues and have not heard from anyone so far.

Just looking at your picture alone I can't tell what could have happened.  I hope at least you did not power the power the board up while sitting on top of the power supply chassis with nothing in between.

ril3y, 2016-10-23:
@Keith-W this is NOT a forum for support.  This is to post issues with the code base.
http://www.synthetos.com/forum/
That is the support forum.

closed 2016-10-23 by ril3y

harolddmorales, 2016-10-25:
Hi Keith,

I did not figure out the issue (I think it was DOA) but I was able to contact Synthetos, return the product and get a refund. I found an alternative method of for achieving our goals.

Thank you,

Harold Morales
Technical Assistant
SHoP Architects
p 212.889.9005 x385
d 646.798.3319

This message, together with any attachments, is intended only for the use of the individual or entity to whom it is addressed and may contain information that is privileged and confidential. If you have received this message in error, please notify the original sender, SHoP, by return e-mail and delete the message, along with any attachments, from your computer. Thank you.