synthetos/TinyG #180: Probing with feedrate less than 18mm/sec puts machine into FROZEN state and RESET is needed.

Tegeon, 2016-06-10:
This issue is probably related to #176. I am unable to probe on TinyG with less than 18mm/sec feedrate. I have recorded 2 probing cycles starting with F20, then F19, F18 and F17 resulting in machine freeze, then RESET and another cycle.

You will find the video here: https://youtu.be/0h_I34pYugA

Tegeon, 2016-06-10:
The sjps returns < {"er":{"fb":442.04,"st":31,"msg":"Move time is NAN"}} when faulty probe (F17 or less) is used. I attach the SJPS log below.
[sjps-verbose-log.txt](https://github.com/synthetos/TinyG/files/309793/sjps-verbose-log.txt)

ril3y, 2026-11-05:
This is the "Move time is NAN" error in your log (`{"er":{"fb":442.04,"st":31,"msg":"Move time is NAN"}}`). A slow move that starts from rest can come out of the planner with no acceleration or deceleration section, and the planner then set its cruise velocity to the entry velocity, which is zero from rest; the move time became infinite and the board stopped. Below roughly F18 at the default jerk, a probe move falls into that case.

It is fixed on edge-next since build 450.01 (https://github.com/synthetos/TinyG/commit/6e4cd1b3) and in the current build, 450.02. We checked it on hardware with a slow move from rest at low jerk ($xjm=5000, F5), which stops with exactly this error when the fix is reverted and completes with it. Probing and homing moves go through the same planner code, so a G38.2 at F15 is covered by the same change. Master (440.21) still has the old code.

Release: https://github.com/synthetos/TinyG/releases/tag/450.02

closed 2026-11-05 by ril3y