synthetos/TinyG #181: Helical bore = end mill plunge = fail

UltiBots, 2016-06-14:
Helical bore = end mill plunge = fail

We are machining 2 parts in 6.5mm aluminum on a desktop CNC with F360 & CAM > Chilipeppr > Tinyg controller. The V4 design has 3.02mm holes and they bored perfectly with a 3/32" single flute end mill. With the V7 design (identical holes and bore settings) the first 3 hole PLUNGE but the subsequent holes results in normal helical bores. Eventually the chips fuse to the end mill and the end mill breaks. 

This behavior has been duplicated on 2 different GCODE senders (Chilipeppr & UniversalGcodeSender) BUT... the bore movements look correct in their respective viewers.

The problem is either with F360 CAM + Post, GCODE sender, or TinyG firmware. So far I have eliminated the GCODE sender as the problem. The plunge failure occurs with 2 different sender programs (Chilipeppr & UniversalGcodeSender). See also [F360 forum post](https://forums.autodesk.com/t5/computer-aided-machining-cam/bore-milling-gcode-deep-bit-fusing-plunge/m-p/6381064/highlight/false#M12943).

Attached are 2 design files and their related CAM GCODE. The V4 files bore holes OK and V7 has a mix of bore and plunge. What is odd is they are in the same CAM operation. Same hole size, end mill size, feeds and speeds.

[dmbe_test.zip](https://github.com/synthetos/TinyG/files/314922/dmbe_test.zip)
 
TIA - Brad
brad )at( ultibots dotcom
772-257-one thousand (9am - 7pm Eastern)

UltiBots, 2016-06-14:
Hi All -

A F360 expert elite wrote the following:

> Below is the code from the v7 file backploted with 2 different backploters, both show all the holes, all the helical moves are in the file. I think you are going to have to get the support for TinyG to look at the file at tell you what the control doesn't like because the code would run on my machines fine. If you need some sort of mod to the post processor to make it run on your control it can be done but as the code has no obvious faults I don't think anyone here can help

Pretty sure the finger is pointed at TinyG. I'm not sure but would love to get this resolved.

Thanks again for looking at this - Brad

[image]

[image]

UltiBots, 2016-06-15:
Looks like issue [174](https://github.com/synthetos/TinyG/issues/174) is still a problem.

Currently running the following:
[fb]  firmware build            440.20
[fv]  firmware version            0.97
[hp]  hardware platform           1.00
[hv]  hardware version            8.00
[id]  TinyG ID                    3X3566-KXJ

Thanks for your help with this - Brad

UltiBots, 2016-06-15:
I think I am on to something.

In looking at GCODE in a text editor. Holes that plunge are missing 3 digit decimal precision whereas holes that bore correctly have 3 digit precision. Looks like a very simple fix!!! 

[image]
[image]

Will find/replace offending lines in GCODE and test.

UltiBots, 2016-06-18:
Quick update... A big thanks to [Mark Hughes](http://forums.autodesk.com/t5/user/viewprofilepage/user-id/2782855) on the F360 forum. He provided a [post processor](http://forums.autodesk.com/t5/computer-aided-machining-cam/bore-milling-gcode-deep-bit-fusing-plunge/td-p/6381064/page/2) work around that convert arcs to straight lines. This allowed me to bore the holes needed!!!

In examining the F360 TinyG post processor in a text editor the workaround is a simple as changing line 30 `allowHelicalMoves = true;` to `allowHelicalMoves = false;`

dustinman, 2016-11-17:
Just wanted to confirm that I see the same behavior.  I set up my CNC to drill 121 holes, into my spoilboard  for threaded inserts, using the bore command in fusion 360.  At the time I had allowHelicalMoves = true; set (I haven't tried it again since I switched to false).  Each hole was different.  Some of them were perfect, others the bit would plunge straight down, then start moving, others it would helical down perfectly, then move off in another direction randomly.  It pretty much ruined my spoil board 👎.  Some of my threaded inserts ended up being too small for the holes and wouldn't stay in.  In the end, the messed up holes were all evenly spaced, but I couldn't find a pattern to when it would or wouldn't work.

engiadina, 2017-03-13:
I have added a file which fails completely to be processed correct. All Circles are NOT executed. Maybe this helps a bit to figure out the problem.
[Bohrung.cnc.zip](https://github.com/synthetos/TinyG/files/838833/Bohrung.cnc.zip)

UltiBots, 2017-03-13:
Hi All - Gave up on TinyG. Will try a Smoothieboard next - Brad

GaylordLivingston, 2017-03-14:
Yup I did the same thing. Without a viable GUI software for set up and operation I was spending all my time working on it instead of getting work done. No Fun. Anybody want to buy one…cheap?

Gaylord R. Livingston

dougkwilson, 2017-03-14:
> Filter service subscribers can train this email as spam or not-spam here: http://my.email-as.net/spamham/cgi-bin/learn.pl?messageid=BCF0696A08C911E7A8A0384FD000F576

Probably.  How much?
  
    Best regards,
  
 Doug Wilson
 250-575-5656

  
 -------- Original Message --------

colinluthier, 2017-10-22:
I beat my head against this problem for the last couple of days. Modifying the post processor in Fusion to only output 3 digits of precision worked for me:

var xyzFormat = createFormat({decimals:3, forceDecimal:true});
var rFormat = xyzFormat; // radius
var abcFormat = createFormat({decimals:3, forceDecimal:true, scale:DEG});
var feedFormat = createFormat({decimals:(unit == MM ? 0 : 1), forceDecimal:true});

colinluthier, 2017-10-22:
I should note that I am using Fusion 360>Chillipepprtablet>TinyG and I work in inches.

[fb]  firmware build            440.20
[fv]  firmware version            0.97
[hp]  hardware platform           1.00
[hv]  hardware version            8.00

colinluthier, 2017-10-24:
Fusion released a new post processor today that eliminated any helical commands and switched everything from G53 to G28. It solved a couple of problems at the expense of G2/G3 moves.

ril3y, 2021-04-10:
Closing due to inactivity.

closed 2021-04-10 by ril3y