synthetos/TinyG #182: G2/G3 full circle not working

MrNobug, 2016-06-18:
I'm not getting any move at all when trying a full circle with G2 and G3 commands. Using 420.20 version.
Seems to be similar problem as earlier: 

339.09: BUG FIXES IN 0.94
•Issue #14 - Full circle G02 results in error - fixed

Error message: "Move less than minimum length"

aldenhart, 2016-06-19:
Can you please post the Gcode you are using? Not just the offending line, but those leading up to it? Thanks

MrNobug, 2016-06-19:
Hi,

The following code generates the error message and there are no arc 
moves. (The G0 blocks work)

G0 X10 Y0
G2 X10 Y0 i-5

G3 X10 Y0 i-5

G0 X10 Y10
G2 X10 Y10 j-5

The following semi circle works fine
G0 X10 Y0
G2 X20 Y0 i5

Den 2016-06-19 kl. 02:03, skrev Alden Hart:

plastik6, 2016-09-03:
Same issue for me, some G3 lines are skipped.
Gcode file generated with fusion360 and run with chilipeppr

[testPocheLead06.zip](https://github.com/synthetos/TinyG/files/453309/testPocheLead06.zip)

madeinoz67, 2016-09-04:
I'm also having issues with Fusion post processed using TinyG post processor as supplied in Fusion .  Am using current stable 440.20, Have tried edge 446.19, this is working but is not reporting machine position under Chilipeppr.  I'm finding that the attached file is failing on the facing cycle, gcode is correct (from what I can see) but on 440.20 it appears that the arcs are starting at start of X and is ending at end of Y, resulting in a diagonal line.
[FETTLER-tinyg.txt](https://github.com/synthetos/TinyG/files/453815/FETTLER-tinyg.txt)

madeinoz67, 2016-09-05:
Update..I now found  a post file (as outlined in https://github.com/synthetos/TinyG/issues/181 ) that does not use Arcs and works as a temporary workaround, however the planner pauses on all the tiny arc segments or motion is jittery doing arcs, so would love to see this get a fix and pushed out to stable branch.

plastik6, 2016-09-05:
Same issues for me with the post processing that converts arcs into segment: motion is very erratic and the feed speed is not maintained..

ril3y, 2026-11-16:
Full circles (endpoint equal to the start) run as a full revolution in 450.01 and later, for both G2 and G3. The skipped-arc cases from the attached Fusion files were fixed by the new arc code (see #240).

Fix: https://github.com/synthetos/TinyG/commit/35814620 (first released in [450.01](https://github.com/synthetos/TinyG/releases/tag/450.01))
Regression test: `tests/hil/tests/bugs/test_e017_ccw_full_circle_endpoint.py`

closed 2026-11-16 by ril3y