synthetos/TinyG #184: how to compile tinyg with atxmega128a1u have external 32k sram

shamsiqbal, 2016-07-19:
hi i am trying to compile tinyg project in atmel studio 7 and i have tried to change axmega192a3 chip with atxmega128a1u but it give some error about vport while compiling with atxmega12a1 (non usb version) gives memory overflow error so i am trying to compile this project with external 32k sram tried many setting described in this link http://www.avrfreaks.net/forum/how-configure-atmel-studio-external-memory-sram and until now i am not successful to compile the project without errors or memory overflow error 
      at the moment i am choosing this chip because it have more i/o's available and i have a quantity available in my stock i hope i will be get some help to compile the project with atxmega128a1/u chip using external 32k sram chip
waiting for asp response 
regards
Shams

ril3y, 2017-06-30:
Sorry outside of scope.  We do not support this cpu / external memory.

closed 2017-06-30 by ril3y

shamsiqbal, 2017-07-01:
hi
i had contacted with atmel and after that i found out that there were some bugs in  avr studio 7 and they have resolved the issue with compiling 
well thank you for very early reply i will be working on grbl as i have added dc brushed servo motor support in it 
if you guys are interested to add dc brushed servo motor support then plz guide me