synthetos/TinyG #186: M6 and touch plate problems

TheOncomingStorm, 2016-09-11:
What is the status of the M6 function thus far? I'm having an issue where I can't use the ChiliPeppr touch plate widget when my program calls an M6. This is in more detail here:

https://www.synthetos.com/topics/hangs-due-to-attempted-use-of-touch-plate-during-program/#post-9908

Apparently this has to do with issue 75.  My tool change macro is simple and the only thing it calls that affects this is the M6 command. Once it does that if I try to use the touch plate widget, TinyG locks up and becomes unresponsive.

TheOncomingStorm, 2016-11-24:
Anyone?

ril3y, 2016-11-25:
Can you give your exact steps to replicate this issue outside of Chilipeppr.  If I follow your information its when you issue an M6 in the middle of a program?

TheOncomingStorm, 2016-11-25:
Hi, yes, that's essentially it. I run a program that has tool changes in it, and when it issues an M6, if I try to do use a, it locks up the TinyG and typically won't respond again until I press the reset button. So it goes like this...

Run program until M6. 
Attempt to do a G38.2. 
TinyG locks.

chilipeppr, 2016-11-25:
I have not seen this happen myself, although I haven't issued a G38.2 after a tool change in a while, but what ChiliPeppr does on an M6 is simply not send any of the Gcode file after the M6 line. It essentially treats the M6 line as if it is the end of the Gcode file. Then when it sees the TinyG say it executed that M6 line it pops up the window in ChiliPeppr saying "ok, m6 pause". Then you can do whatever you want in ChiliPeppr. Then, when you unpause the Gcode widget it just starts playing the file again from the line after the M6. So, if you think about it, ChiliPeppr does pretty much nothing at all because it's just treating the Gcode file as if it is multiple files.

What would help on this bug better, is to show the actual Gcode from the Serial Port Console in ChiliPeppr that you do after the M6. My gut tells me this has nothing to do with M6 at all and simply is a crash on the G38.2 which I do recall seeing in older firmware.

BTW, I ran a ton of G38.2's over the last week and they worked like a champ.

TheOncomingStorm, 2016-11-25:
Ok, I've been away from the CNC for a while so I'll have to head down there and run one of the programs that was causing it. What is the best way to capture the gcode from the serial port console, I'm guessing a simple copy/paste, or is there a log or save function?

G38.2 works by itself just find as far as I can tell. It seems to only be the combination of using it with M6 that causes the problem but that doesn't necessarily mean that M6 is the cause... Give me a day to churn something out. I'll make a really simple program so we don't have to deal with the full project I was working on.

ril3y, 2016-11-26:
Run spjs via verbose? John thoughts?

johnlauer, 2016-11-26:
Nah. That won’t help. I bet tinyg gives an error, but it’s hidden in serial port console unless you turn off the filter (funnel) icon.

TheOncomingStorm, 2016-12-13:
Sorry for the delay. I think I removed the filter(it's hard to tell...no color change or the like to the filter button) but regardless I tried it, then pressed the filter button again and tried it again, both gave me the same output. 

http://pastebin.com/71MzU1Dy

chilipeppr, 2016-12-13:
I tHink ChiliPeppr just looks for m6 not m06. That could be it. That's a
small code change.

TheOncomingStorm, 2016-12-13:
I probably also had this as M6 as well. I changed the tool change macro several times to debug, and I'm pretty sure that I changed from M6 to M06 when I was reading a page on Gcode syntax.

chilipeppr, 2016-12-13:
My gut tells me this is a TinyG bug then. G38.2 has crashed on me in the past in older firmware versions.

ril3y, 2026-11-16:
The lockup reported here (G38.2 right after a tool change) matches problems fixed in 450.01: JSON-wrapped G-code could overtake a pending probe or queued commands, a block queuing several commands could stop the board with "Buffer full fatal", and a probe could start before earlier moves finished.

A hardware regression test now runs `T3 M6` (and `T4 M6` sent as JSON) followed by two G38.2 probes: each tool change is accepted, every probe runs its full travel and reports, and the board stays responsive. M6 itself only makes the T tool current; it does not pause (see the Gcode-Support wiki page).

Test: [tests/hil/tests/bugs/test_i186_probe_after_tool_change.py](https://github.com/synthetos/TinyG/blob/03cc86e5/tests/hil/tests/bugs/test_i186_probe_after_tool_change.py) (passes on 450.06; edge-next 03cc86e5). Please reopen if you still see a hang on 450.06.

closed 2026-11-16 by ril3y