synthetos/TinyG #187: COOL PWM SDIR SPIN detailed description

juri74, 2016-09-19:
hello i need detailed description of pin COOL, PWM, SDIR and SPIN
there are any information at all at the page "Connecting TinyG"
and i can't find anything useful.
i have a spindle driver with input requirement of level 3.5-12V VPP, frequency 1K-5KHZ , duty cycle of 0% -93%. 
can i use it?

cmcgrath5035, 2016-09-28:
Coolant toggles with Gcodes M7, M8 and M9. See https://github.com/synthetos/TinyG/wiki/Gcode-Support
SDIR toggles with Gcodes M3, M4 (direction)
SPIN is toggle With M3/M4 spindle On/Off logic
PWM is active based on settings - see https://dl.dropboxusercontent.com/u/50261731/Spindle%20Test%201.doc

juri74, 2016-09-28:
thank you for your reply, 
i read and now i know how to control the outputs via M codes,
but now the problem is how to connect the outputs! 
example for the coolant i have this pump (manualy driven actually) http://www.ebay.it/itm/3-5M-75W-3500L-H-Engraving-Machine-Cooling-Water-Pump-For-CNC-Spindle-Motor-/162212141310?hash=item25c498d8fe:g:H2AAAOSwmfhX4tUD
this is a spindle coolant pump but i use it as coolant pump
how i connect it to TinyG?

cmcgrath5035, 2016-09-29:
tinyG outputs are unbuffered 3V CMOS logic.
Send an M3 command and SPIN lead will toggle to 3V from 0V and PWM will output a PWM pulse stream at selected frequency as defined by the calibration $p1* parameters. 
An M5 command will return SPIN to 0V, PWM to whatever PWM duty cycle is set by $p1pof.

You probably need some sort of logic to power on/off circuit to control your hardware.
There are all sorts of interfaces available, depending on the load current required.

juri74, 2016-09-29:
ok thank you again
now is more clear! i surely need a little interface, no problem that's everyday work :)
this can be closed but i suggest to put these infos on wiki pages
thanks again

ril3y, 2026-11-16:
The outputs are unbuffered 3.3 V logic: SPIN and SDIR follow M3/M4/M5, COOL follows M7/M8/M9, and PWM follows the `$p1` settings (polarity via `$spep`/`$spdp`). Loads need an external driver or relay board.

closed 2026-11-16 by ril3y