synthetos/TinyG #188: Confusing status report data for continuous linear move over multiple GCodes

jugglingcats, 2016-09-27:
Hi, I am looking to create a simple DRO using the TinyG JSON API directly.

I have configured TinyG to send a status report every 100ms using `{si:100}`.

I then issue some GCode:
```
{gc: "g64"}
{gc: "g1x25f500"}
{gc: "g1x50f500"}
```

I am expecting this to run continuously at the feedrate through the junction, but it seems to slow down to a complete stop instead.

The status reports do not have any timecode on them but my assumption is they will be at or close to 100ms intervals throughout the move. Is this a valid assumption?

Many thanks for your help.

jugglingcats, 2016-09-27:
Looks like this is maybe a buffering issue: if I add more than two commands the final two are planned correctly, but the motor stops after the first g1.

For example `g1x50` `g1x75` `g1x100` gives a pause at X=50 but is continuous thereafter.

Is this expected behaviour?

aldenhart, 2016-09-27:
Are you sending these in a single line or are these multiple lines? Gcode requires that they be on individual lines (Gcode blocks). Presumably you sent a feed rate some time before this or that sequence will not work.

Separately, ate you referring to the status reports pausing, or the motion itself?

jugglingcats, 2016-09-27:
Thanks for your interest. To clarify I am not hooked up to motors at this point, so am just sending commands and receiving status reports. All settings are default.

In both scenarios I am sending g1 moves with feedrate specified on each move (I know this is not strictly necessary). I am then looking at the status report output, and also ended up graphing it for good measure. There are newlines being sent after each JSON line in the examples below.

Setup:
```
{sv:2}
{si:100}
{gc:"g64"}
{gc:"g0x0"}
```

Scenario 1: Two moves
```
{gc:"g1x25f500"}
{gc:"g1x50f500"}
```
(graph shows velocity at zero at x=25)

Scenario 1: Three moves
```
{gc:"g1x25f500"}
{gc:"g1x50f500"}
{gc:"g1x75f500"}
```
(graph shows velocity at zero at x=25 and cruise velocity at x=50)

aldenhart, 2016-09-27:
Here's what I get when I do that. This looks OK to me - are you getting something different?

<pre>
{"r":{"sv":2},"f":[1,0,7]}
{"r":{"si":100},"f":[1,0,9]}
{"r":{},"f":[1,0,11]}
{"r":{},"f":[1,0,12]}
{"r":{},"f":[1,0,17]}
{"r":{},"f":[1,0,17]}
{"r":{},"f":[1,0,17]}
{"sr":{"posx":0.079,"posy":0,"posz":0,"posa":0,"line":0,"vel":221.08,"feed":500,"stat":5,"macs":5,"cycs":1,"mots":2,"hold":0,"dist":0,"admo":1}}
{"sr":{"posx":0.813,"posy":0,"posz":0,"posa":0,"line":0,"vel":500,"feed":500,"stat":5,"macs":5,"cycs":1,"mots":2,"hold":0,"dist":0,"admo":1}}
{"sr":{"posx":1.638,"posy":0,"posz":0,"posa":0,"line":0,"vel":500,"feed":500,"stat":5,"macs":5,"cycs":1,"mots":2,"hold":0,"dist":0,"admo":1}}
{"sr":{"posx":2.476,"posy":0,"posz":0,"posa":0,"line":0,"vel":500,"feed":500,"stat":5,"macs":5,"cycs":1,"mots":2,"hold":0,"dist":0,"admo":1}}
{"sr":{"posx":3.313,"posy":0,"posz":0,"posa":0,"line":0,"vel":500,"feed":500,"stat":5,"macs":5,"cycs":1,"mots":2,"hold":0,"dist":0,"admo":1}}
{"sr":{"posx":4.151,"posy":0,"posz":0,"posa":0,"line":0,"vel":500,"feed":500,"stat":5,"macs":5,"cycs":1,"mots":2,"hold":0,"dist":0,"admo":1}}
{"sr":{"posx":4.988,"posy":0,"posz":0,"posa":0,"line":0,"vel":500,"feed":500,"stat":5,"macs":5,"cycs":1,"mots":2,"hold":0,"dist":0,"admo":1}}
{"sr":{"posx":5.826,"posy":0,"posz":0,"posa":0,"line":0,"vel":500,"feed":500,"stat":5,"macs":5,"cycs":1,"mots":2,"hold":0,"dist":0,"admo":1}}
{"sr":{"posx":6.663,"posy":0,"posz":0,"posa":0,"line":0,"vel":500,"feed":500,"stat":5,"macs":5,"cycs":1,"mots":2,"hold":0,"dist":0,"admo":1}}
{"sr":{"posx":7.501,"posy":0,"posz":0,"posa":0,"line":0,"vel":500,"feed":500,"stat":5,"macs":5,"cycs":1,"mots":2,"hold":0,"dist":0,"admo":1}}
{"sr":{"posx":8.338,"posy":0,"posz":0,"posa":0,"line":0,"vel":500,"feed":500,"stat":5,"macs":5,"cycs":1,"mots":2,"hold":0,"dist":0,"admo":1}}
{"sr":{"posx":9.176,"posy":0,"posz":0,"posa":0,"line":0,"vel":500,"feed":500,"stat":5,"macs":5,"cycs":1,"mots":2,"hold":0,"dist":0,"admo":1}}
{"sr":{"posx":10.013,"posy":0,"posz":0,"posa":0,"line":0,"vel":500,"feed":500,"stat":5,"macs":5,"cycs":1,"mots":2,"hold":0,"dist":0,"admo":1}}
{"sr":{"posx":10.85,"posy":0,"posz":0,"posa":0,"line":0,"vel":500,"feed":500,"stat":5,"macs":5,"cycs":1,"mots":2,"hold":0,"dist":0,"admo":1}}
{"sr":{"posx":11.688,"posy":0,"posz":0,"posa":0,"line":0,"vel":500,"feed":500,"stat":5,"macs":5,"cycs":1,"mots":2,"hold":0,"dist":0,"admo":1}}
{"sr":{"posx":12.525,"posy":0,"posz":0,"posa":0,"line":0,"vel":500,"feed":500,"stat":5,"macs":5,"cycs":1,"mots":2,"hold":0,"dist":0,"admo":1}}
{"sr":{"posx":13.363,"posy":0,"posz":0,"posa":0,"line":0,"vel":500,"feed":500,"stat":5,"macs":5,"cycs":1,"mots":2,"hold":0,"dist":0,"admo":1}}
{"sr":{"posx":14.2,"posy":0,"posz":0,"posa":0,"line":0,"vel":500,"feed":500,"stat":5,"macs":5,"cycs":1,"mots":2,"hold":0,"dist":0,"admo":1}}
{"sr":{"posx":15.038,"posy":0,"posz":0,"posa":0,"line":0,"vel":500,"feed":500,"stat":5,"macs":5,"cycs":1,"mots":2,"hold":0,"dist":0,"admo":1}}
{"sr":{"posx":15.875,"posy":0,"posz":0,"posa":0,"line":0,"vel":500,"feed":500,"stat":5,"macs":5,"cycs":1,"mots":2,"hold":0,"dist":0,"admo":1}}
{"sr":{"posx":16.713,"posy":0,"posz":0,"posa":0,"line":0,"vel":500,"feed":500,"stat":5,"macs":5,"cycs":1,"mots":2,"hold":0,"dist":0,"admo":1}}
{"sr":{"posx":17.55,"posy":0,"posz":0,"posa":0,"line":0,"vel":500,"feed":500,"stat":5,"macs":5,"cycs":1,"mots":2,"hold":0,"dist":0,"admo":1}}
{"sr":{"posx":18.388,"posy":0,"posz":0,"posa":0,"line":0,"vel":500,"feed":500,"stat":5,"macs":5,"cycs":1,"mots":2,"hold":0,"dist":0,"admo":1}}
{"sr":{"posx":19.225,"posy":0,"posz":0,"posa":0,"line":0,"vel":500,"feed":500,"stat":5,"macs":5,"cycs":1,"mots":2,"hold":0,"dist":0,"admo":1}}
{"sr":{"posx":20.063,"posy":0,"posz":0,"posa":0,"line":0,"vel":500,"feed":500,"stat":5,"macs":5,"cycs":1,"mots":2,"hold":0,"dist":0,"admo":1}}
{"sr":{"posx":20.9,"posy":0,"posz":0,"posa":0,"line":0,"vel":500,"feed":500,"stat":5,"macs":5,"cycs":1,"mots":2,"hold":0,"dist":0,"admo":1}}
{"sr":{"posx":21.737,"posy":0,"posz":0,"posa":0,"line":0,"vel":500,"feed":500,"stat":5,"macs":5,"cycs":1,"mots":2,"hold":0,"dist":0,"admo":1}}
{"sr":{"posx":22.575,"posy":0,"posz":0,"posa":0,"line":0,"vel":500,"feed":500,"stat":5,"macs":5,"cycs":1,"mots":2,"hold":0,"dist":0,"admo":1}}
{"sr":{"posx":23.412,"posy":0,"posz":0,"posa":0,"line":0,"vel":500,"feed":500,"stat":5,"macs":5,"cycs":1,"mots":2,"hold":0,"dist":0,"admo":1}}
{"sr":{"posx":24.25,"posy":0,"posz":0,"posa":0,"line":0,"vel":500,"feed":500,"stat":5,"macs":5,"cycs":1,"mots":2,"hold":0,"dist":0,"admo":1}}
{"sr":{"posx":25.088,"posy":0,"posz":0,"posa":0,"line":0,"vel":500,"feed":500,"stat":5,"macs":5,"cycs":1,"mots":2,"hold":0,"dist":0,"admo":1}}
{"sr":{"posx":25.925,"posy":0,"posz":0,"posa":0,"line":0,"vel":500,"feed":500,"stat":5,"macs":5,"cycs":1,"mots":2,"hold":0,"dist":0,"admo":1}}
{"sr":{"posx":26.763,"posy":0,"posz":0,"posa":0,"line":0,"vel":500,"feed":500,"stat":5,"macs":5,"cycs":1,"mots":2,"hold":0,"dist":0,"admo":1}}
{"sr":{"posx":27.6,"posy":0,"posz":0,"posa":0,"line":0,"vel":500,"feed":500,"stat":5,"macs":5,"cycs":1,"mots":2,"hold":0,"dist":0,"admo":1}}
{"sr":{"posx":28.438,"posy":0,"posz":0,"posa":0,"line":0,"vel":500,"feed":500,"stat":5,"macs":5,"cycs":1,"mots":2,"hold":0,"dist":0,"admo":1}}
{"sr":{"posx":29.275,"posy":0,"posz":0,"posa":0,"line":0,"vel":500,"feed":500,"stat":5,"macs":5,"cycs":1,"mots":2,"hold":0,"dist":0,"admo":1}}
[... cut ...]
{"sr":{"posx":74.449,"posy":0,"posz":0,"posa":0,"line":0,"vel":499.78,"feed":500,"stat":5,"macs":5,"cycs":1,"mots":2,"hold":0,"dist":0,"admo":1}}
{"sr":{"posx":74.989,"posy":0,"posz":0,"posa":0,"line":0,"vel":58.79,"feed":500,"stat":5,"macs":5,"cycs":1,"mots":2,"hold":0,"dist":0,"admo":1}}
{"sr":{"posx":75,"posy":0,"posz":0,"posa":0,"line":0,"vel":0,"feed":500,"stat":3,"macs":3,"cycs":0,"mots":0,"hold":0,"dist":0,"admo":1}}
{"sr":{"posx":75,"posy":0,"posz":0,"posa":0,"line":0,"vel":0,"feed":500,"stat":3,"macs":3,"cycs":0,"mots":0,"hold":0,"dist":0,"admo":1}}
</pre>

jugglingcats, 2016-09-27:
I am running TinyG fv: 0.970, build: 440.20. Below is my output.

Perhaps I should be using g2 going forward.

```
{"r":{"fv":0.970,"fb":440.20,"hp":1,"hv":8,"id":"3Y2462-VUN","msg":"Initializing configs to default settings"},"f":[1,15,14,6629]}
{"sr":{"posx":0.000,"posy":0.000,"posz":0.000,"posa":0.000,"feed":0.00,"vel":0.00,"unit":1,"coor":1,"dist":0,"frmo":1,"stat":1}}
{"r":{"sv":2},"f":[1,0,9,6902]}
{"sr":{"posx":0.000,"posy":0.000,"posz":0.000,"posa":0.000,"feed":0.00,"vel":0.00,"unit":1,"coor":1,"dist":0,"frmo":1,"stat":1}}
{"r":{"si":100},"f":[1,0,11,9547]}
{"sr":{"posx":0.000,"posy":0.000,"posz":0.000,"posa":0.000,"feed":0.00,"vel":0.00,"unit":1,"coor":1,"dist":0,"frmo":1,"stat":1}}
{"r":{"sr":{"posx":0.000,"posy":0.000,"posz":0.000,"posa":0.000,"feed":0.00,"vel":0.00,"unit":1,"coor":1,"dist":0,"frmo":1,"stat":1}},"f":[1,0,12,1192]}
{"sr":{"posx":0.000,"posy":0.000,"posz":0.000,"posa":0.000,"feed":0.00,"vel":0.00,"unit":1,"coor":1,"dist":0,"frmo":1,"stat":1}}
{"r":{},"f":[1,0,13,72]}
{"sr":{"posx":0.000,"posy":0.000,"posz":0.000,"posa":0.000,"feed":0.00,"vel":0.00,"unit":1,"coor":1,"dist":0,"frmo":1,"stat":1}}
{"r":{},"f":[1,0,14,73]}
{"sr":{"posx":0.000,"posy":0.000,"posz":0.000,"posa":0.000,"feed":0.00,"vel":0.00,"unit":1,"coor":1,"dist":0,"frmo":1,"stat":1}}
{"r":{},"f":[1,0,19,78]}
{"sr":{"posx":0.000,"posy":0.000,"posz":0.000,"posa":0.000,"feed":500.00,"vel":0.01,"unit":1,"coor":1,"dist":0,"frmo":1,"stat":5}}
{"r":{},"f":[1,0,19,78]}
{"sr":{"posx":0.000,"posy":0.000,"posz":0.000,"posa":0.000,"feed":500.00,"vel":0.25,"unit":1,"coor":1,"dist":0,"frmo":1,"stat":5}}
{"r":{},"f":[1,0,19,78]}
{"sr":{"posx":0.000,"posy":0.000,"posz":0.000,"posa":0.000,"feed":500.00,"vel":1.11,"unit":1,"coor":1,"dist":0,"frmo":1,"stat":5}}
{"sr":{"posx":0.024,"posy":0.000,"posz":0.000,"posa":0.000,"feed":500.00,"vel":37.59,"unit":1,"coor":1,"dist":0,"frmo":1,"stat":5}}
{"sr":{"posx":0.162,"posy":0.000,"posz":0.000,"posa":0.000,"feed":500.00,"vel":141.05,"unit":1,"coor":1,"dist":0,"frmo":1,"stat":5}}
{"sr":{"posx":0.527,"posy":0.000,"posz":0.000,"posa":0.000,"feed":500.00,"vel":292.73,"unit":1,"coor":1,"dist":0,"frmo":1,"stat":5}}
{"sr":{"posx":1.104,"posy":0.000,"posz":0.000,"posa":0.000,"feed":500.00,"vel":421.19,"unit":1,"coor":1,"dist":0,"frmo":1,"stat":5}}
{"sr":{"posx":1.878,"posy":0.000,"posz":0.000,"posa":0.000,"feed":500.00,"vel":491.20,"unit":1,"coor":1,"dist":0,"frmo":1,"stat":5}}
{"sr":{"posx":2.667,"posy":0.000,"posz":0.000,"posa":0.000,"feed":500.00,"vel":500.00,"unit":1,"coor":1,"dist":0,"frmo":1,"stat":5}}
{"sr":{"posx":3.500,"posy":0.000,"posz":0.000,"posa":0.000,"feed":500.00,"vel":500.00,"unit":1,"coor":1,"dist":0,"frmo":1,"stat":5}}
{"sr":{"posx":4.292,"posy":0.000,"posz":0.000,"posa":0.000,"feed":500.00,"vel":500.00,"unit":1,"coor":1,"dist":0,"frmo":1,"stat":5}}
{"sr":{"posx":5.125,"posy":0.000,"posz":0.000,"posa":0.000,"feed":500.00,"vel":500.00,"unit":1,"coor":1,"dist":0,"frmo":1,"stat":5}}
{"sr":{"posx":5.917,"posy":0.000,"posz":0.000,"posa":0.000,"feed":500.00,"vel":500.00,"unit":1,"coor":1,"dist":0,"frmo":1,"stat":5}}
{"sr":{"posx":6.750,"posy":0.000,"posz":0.000,"posa":0.000,"feed":500.00,"vel":500.00,"unit":1,"coor":1,"dist":0,"frmo":1,"stat":5}}
{"sr":{"posx":7.542,"posy":0.000,"posz":0.000,"posa":0.000,"feed":500.00,"vel":500.00,"unit":1,"coor":1,"dist":0,"frmo":1,"stat":5}}
{"sr":{"posx":8.375,"posy":0.000,"posz":0.000,"posa":0.000,"feed":500.00,"vel":500.00,"unit":1,"coor":1,"dist":0,"frmo":1,"stat":5}}
{"sr":{"posx":9.167,"posy":0.000,"posz":0.000,"posa":0.000,"feed":500.00,"vel":500.00,"unit":1,"coor":1,"dist":0,"frmo":1,"stat":5}}
{"sr":{"posx":10.000,"posy":0.000,"posz":0.000,"posa":0.000,"feed":500.00,"vel":500.00,"unit":1,"coor":1,"dist":0,"frmo":1,"stat":5}}
{"sr":{"posx":10.792,"posy":0.000,"posz":0.000,"posa":0.000,"feed":500.00,"vel":500.00,"unit":1,"coor":1,"dist":0,"frmo":1,"stat":5}}
{"sr":{"posx":11.625,"posy":0.000,"posz":0.000,"posa":0.000,"feed":500.00,"vel":500.00,"unit":1,"coor":1,"dist":0,"frmo":1,"stat":5}}
{"sr":{"posx":12.417,"posy":0.000,"posz":0.000,"posa":0.000,"feed":500.00,"vel":500.00,"unit":1,"coor":1,"dist":0,"frmo":1,"stat":5}}
{"sr":{"posx":13.250,"posy":0.000,"posz":0.000,"posa":0.000,"feed":500.00,"vel":500.00,"unit":1,"coor":1,"dist":0,"frmo":1,"stat":5}}
{"sr":{"posx":14.042,"posy":0.000,"posz":0.000,"posa":0.000,"feed":500.00,"vel":500.00,"unit":1,"coor":1,"dist":0,"frmo":1,"stat":5}}
{"sr":{"posx":14.875,"posy":0.000,"posz":0.000,"posa":0.000,"feed":500.00,"vel":500.00,"unit":1,"coor":1,"dist":0,"frmo":1,"stat":5}}
{"sr":{"posx":15.667,"posy":0.000,"posz":0.000,"posa":0.000,"feed":500.00,"vel":500.00,"unit":1,"coor":1,"dist":0,"frmo":1,"stat":5}}
{"sr":{"posx":16.500,"posy":0.000,"posz":0.000,"posa":0.000,"feed":500.00,"vel":500.00,"unit":1,"coor":1,"dist":0,"frmo":1,"stat":5}}
{"sr":{"posx":17.292,"posy":0.000,"posz":0.000,"posa":0.000,"feed":500.00,"vel":500.00,"unit":1,"coor":1,"dist":0,"frmo":1,"stat":5}}
{"sr":{"posx":18.125,"posy":0.000,"posz":0.000,"posa":0.000,"feed":500.00,"vel":500.00,"unit":1,"coor":1,"dist":0,"frmo":1,"stat":5}}
{"sr":{"posx":18.958,"posy":0.000,"posz":0.000,"posa":0.000,"feed":500.00,"vel":500.00,"unit":1,"coor":1,"dist":0,"frmo":1,"stat":5}}
{"sr":{"posx":19.750,"posy":0.000,"posz":0.000,"posa":0.000,"feed":500.00,"vel":500.00,"unit":1,"coor":1,"dist":0,"frmo":1,"stat":5}}
{"sr":{"posx":20.583,"posy":0.000,"posz":0.000,"posa":0.000,"feed":500.00,"vel":500.00,"unit":1,"coor":1,"dist":0,"frmo":1,"stat":5}}
[... cut ...]
{"sr":{"posx":74.987,"posy":0.000,"posz":0.000,"posa":0.000,"feed":500.00,"vel":27.39,"unit":1,"coor":1,"dist":0,"frmo":1,"stat":5}}
{"sr":{"posx":75.000,"posy":0.000,"posz":0.000,"posa":0.000,"feed":500.00,"vel":0.12,"unit":1,"coor":1,"dist":0,"frmo":1,"stat":5}}
{"sr":{"posx":75.000,"posy":0.000,"posz":0.000,"posa":0.000,"feed":500.00,"vel":0.00,"unit":1,"coor":1,"dist":0,"frmo":1,"stat":3}}
{"sr":{"posx":75.000,"posy":0.000,"posz":0.000,"posa":0.000,"feed":500.00,"vel":0.00,"unit":1,"coor":1,"dist":0,"frmo":1,"stat":3}}
```

jugglingcats, 2016-09-27:
I have a v8 board so from what I understand it is not possible to run g2 on it right now?

aldenhart, 2016-09-27:
My mistake. I think I know what's going on. On v8 the first move in plans from zero to zero. The rest of the moves plan from zero and continue. It's a planner stutter. If you put a really short move in before the "real" moves this becomes negligible. It's something we've been planning to fix on v8. 

g2 will not run on the v8s. We are working on new hardware that will run this.

jugglingcats, 2016-09-27:
That's clear. I may try g2 on Arduino Due for comparison. Many thanks for the info!

closed 2016-09-28 by aldenhart