synthetos/TinyG #189: G2/G3 wont create full circle

cago6586, 2016-09-28:
Hello,

 I have been having issues with the G2/G3 arc commands. When I try to make a full circle the motors seem to get power but do not move, and coolterm gives an error: Move less than minimum length.

I've tried circles centered on the origin, as well as in different quadrants. The only way I can get a full circle to work is by splitting it up into four arc segments. 

Firmware version is 0.97 and build is 440.20 

Gcode for Full circle around origin:

G2 X10Y0 I-10J0

Gcode for four arc segments around origin: 

G2 X0Y-10 I-10J0
G2 X-10Y0 I0J10
G2 X0Y10 I10J0
G2 X10Y0 I0J-10

Any help is appreciated. 

Thanks,
Carmelo

troisiemetype, 2016-09-28:
I don't know if it's an "intented" or real issue. I think that when you enter a coordinate (i.e. a place to move to), the planner computes the delta between the current position and the goal position. In case of a complete circle, these two position are the same one, so the planner must be "fooled" and think it doesn't have to create any move.
When I need a complete circle, I split it in two. Something like this:

G90
G0 X10 Y10
G2 X30 R10
G2 X10 R10

The R parameter is convenient: it permits not to have to figure where is the center when coding simple program by hand...

Hope it helps,

Pierre-Loup.

ril3y, 2026-11-16:
Fixed in 450.01: a single G2 or G3 whose endpoint equals the start now runs a full circle instead of reporting "Move less than minimum length".

Fix: https://github.com/synthetos/TinyG/commit/35814620 (first released in [450.01](https://github.com/synthetos/TinyG/releases/tag/450.01))
Regression test: `tests/hil/tests/bugs/test_e017_ccw_full_circle_endpoint.py`

closed 2026-11-16 by ril3y