synthetos/TinyG #197: after feedhold, cycle start sometimes causes movement that doesn't stop

greener1, 2017-03-29:
I'm running : Firmware Version : 0.97, Firmware Build : 446.19, Hardware Version : 8
I'm using the edge branch because I need the transaction IDs.

After sending a G-code fragment for a relative move, then sending a feedhold (causing the motion to stop as expected), and then sending a cycle start, the movement resumes (as expected) but often goes past the point requested in the original G-code, and in fact just keeps going.  For example:

```
received: {"sr":{"posy":-127.282,"posz":501.442,"vel":0.00,"coor":1,"momo":4,"stat":3}}
sending: {"gc":"g91g1y5f300",tid:33}
received: {"r":{},"tid":33,"f":[3,0,24]}
received: {"sr":{"posy":14.955,"posz":0.702,"vel":300.00,"momo":1,"stat":5}}
received: {"sr":{"posy":16.174}}
received: {"sr":{"posy":17.418}}
received: {"sr":{"posy":18.638}}
sending: !
received: {"sr":{"posy":18.823,"vel":1.67,"stat":6}}
sending: ~
received: {"sr":{"posy":19.996,"vel":300.00,"stat":5}}
received: {"sr":{"posy":21.221}}
received: {"sr":{"posy":22.446}}
received: {"sr":{"posy":23.671}}
received: {"sr":{"posy":24.896}}
received: {"sr":{"posy":26.146}}
```
... continues to ever larger posy, even though it should have stopped at posy ~= 20

Likewise on the Z axis:
```
received: {"sr":{"posy":-60.000,"posz":500.000,"vel":0.00,"stat":3}}
sending: {"gc":"g91g1z-10f300",tid:100}
received: {"r":{},"tid":100,"f":[3,0,24]}
received: {"sr":{"posy":9.121,"posz":3.395,"vel":290.31,"stat":5}}
received: {"sr":{"posz":2.175,"vel":300.00}}
received: {"sr":{"posz":0.952}}
received: {"sr":{"posz":-0.270}}
sending: !
received: {"sr":{"posz":-1.382,"vel":159.53,"stat":6}}
received: {"sr":{"posz":-1.503,"vel":0.00}}
sending: ~
received: {"sr":{"posz":-2.016,"vel":290.31,"stat":5}}
received: {"sr":{"posz":-3.238,"vel":300.00}}
received: {"sr":{"posz":-4.463}}
received: {"sr":{"posz":-5.688}}
received: {"sr":{"posz":-6.938}}
received: {"sr":{"posz":-8.163}}
received: {"sr":{"posz":-9.388}}
received: {"sr":{"posz":-10.613}}
received: {"sr":{"posz":-11.838}}
received: {"sr":{"posz":-13.063}}
received: {"sr":{"posz":-14.288}}
received: {"sr":{"posz":-15.513}}
received: {"sr":{"posz":-16.738}}
```
... again it continues even though only a change in Z by only -10 units was requested.

Am I doing something wrong here?  Sometimes the cycle start after the feedhold does result in the resumed movement only continuing to the requested position.

greener1, 2017-04-01:
By looking at how the Chilipeppr app handles feedhold/resume, successfully.  I see that on resume, in addition to the cycle start command:
`~`
it also sends a queue request:
`{"qr":""}`
When I do that too, the resume works as expected, stopping at the initially requested position.

So maybe the documentation just needs to be updated to show that the resume command is actually a combination of two commands.  As shown in line 691 of the [Chilipeppr code](https://github.com/chilipeppr/widget-tinyg/blob/master/auto-generated-widget.html) it's really:
`'~\n{"qr":""}\n'`

ril3y, 2026-11-16:
Fixed in 450.01: resuming after a feedhold no longer re-runs the whole move from the hold point, so the move stops at its programmed end. A plain `~` is enough; the extra `{"qr":""}` workaround is not needed.

Fix: https://github.com/synthetos/TinyG/commit/1529b9e1, https://github.com/synthetos/TinyG/commit/42e9ac71 (first released in [450.01](https://github.com/synthetos/TinyG/releases/tag/450.01))
Regression test: `tests/hil/tests/bugs/test_e030_resume_overshoot.py`, `tests/hil/tests/bugs/test_n66_accounting_hold_endpoint.py`

closed 2026-11-16 by ril3y