synthetos/TinyG #198: CNC Z-axis upwards movement issue

o700488, 2017-03-30:
Hello TinyG,

I began building a large Open Builds CNC machine a while ago and I almost have it all working, but I'm hung up on this one issue. The z-axis seems to randomly shift upward, increasing the distance from the build at times when it should be maintaining its height. The spindle is quite heavy so it seems especially odd that it shifts up. It shifts a mm or two at a time per every ~250mm which is enough to ruin any of my builds. None of the other axes seem to have this problem. At first I thought it might be electromagnetic interference (from the 2.2kw spindle) but the wires are shielded. The motor makes a bit of a crunching noise when it happens.

Specs:
Tinyg2 controller (Arduino due with 2 GRBL shields). Perhaps its some firmware issue.
Chrome Chilipeppr.com/tinyg?v=9.0
latest firmware.

The current gcode involves, row by row, shaving off layers. (Each pass in the x-axis is a row)

I've tried testing the CNC machine using Coolterm, but whether or not the spindle is on it does NOT shift in the z-axis.
This does not fix the problem as I can not use Coolterm to run the build.

I'm uploading the gcode using Chilipeppr. Maybe that is causing the problem.

I'll do some more testing, but if anyone has any other ideas please let the thread know.

Thank you for your help!

ril3y, 2017-06-30:
@o700488 hey there.  So sending via coolterm does not show the drift?  I am going to loop in @chilipeppr on this one. @jlauer12 thoughts?  Plus if this does turn out to be a tinyg issue its actually a g2 issue and should be opened on there.

chilipeppr, 2017-06-30:
I think it was a work coordinate system issue, but not sure.

ril3y, 2021-04-10:
Closing due to inactivity.

closed 2021-04-10 by ril3y