synthetos/TinyG #199: Non-trivial kinematics

spiderdab, 2017-04-03:
I don't know if it's the right place to ask, but the forum couldn't help. So i try here.

I’m new to g2code, but I already programmed and moved different Linuxcnc machine and some 3d printers with non trivial kinematics.
I’m interested in g2code because I need to prepare a machine with an arduino due, with 4 winch-joints that move wires connected in the middle.
I didn’t find examples, so I’m asking here if the place to insert my changes is the file kinematics.cpp, or if there are other files you can point me to for other changes.

Thanks.

giseburt, 2017-04-03:
Hello,

I believe you're looking for g2core. That's in the repo
https://github.com/synthetos/g2 .

We are about to revamp the kinematics support for just this reason. The
dev-168-gquintic branch has our first stabs at it.
https://github.com/synthetos/g2/blob/dev-168-gquintic/g2core/kinematics.cpp

They're rather clumsy at the moment with the ifdefs and such. Ideally we'll
make the a runtime selection.

Please open another issue in the g2 repo and we can discuss what's needed
to do this properly.

Thank you!
-Rob

closed 2017-04-03 by giseburt

spiderdab, 2017-04-03:
Ops. My distraction..