synthetos/TinyG #202: Sending GCode Commands RealTime

rchisena, 2017-05-03:
Hi all! 

**Background of the Issue**
I am trying to send GCode from a robot controller to the TinyG board to control a stepper motor as the robot moves. As the robot moves through different positions, it populates values to send to a stepper motor. These values include the distance and the speed that the motor should run. 

The robot outputs serial (9-pin, RS-232) with a baudrate of 38400, I have a converter that converts this serial port to USB. I then have a virtual serial port emulator that bridges the robot USB to the TinyG USB. I also have tried a MATLAB program that reads from the robot-USB and sends the GCode value to the TinyG serially. Neither of these paths have worked consistently. 

Configuration for the TinyG board is: 
115200 baud rate, 
CTS Flow Control
No parity bit
1 stop bit

Configuration of Robot: 
38400 baud rate
Flow control - hardware or software
No parity bit, 
1 stop bit

**Issue**
As the robot is outputting commands to the TinyG, the extruder starts to jump as if it is loosing commands. If I run the same Gcode program but from a text file full of the commands, the stepper will work fine, but as soon as I start to run real-time, the board causes the stepper to jump, run backwards, and do just about everything its not supposed to do. 

I have verified that the robot sends the correct commands. It seems there is some communication or timing issue that is occurring in the pass through, but I cannot seem to find out what it is. 

**Request**
If someone has experience with this type of issue or otherwise could provide some helpful ideas, I would really appreciate it. Also,  I will be happy to share what I learn and my research project with the community as well if you guys are interested. 

Best wishes, 

Robert

ril3y, 2017-05-03:
not sure I understand this fully.  I see the baud rate different?  You need
to send and recv at the same rate.

Riley

rchisena, 2017-05-04:
Hi Riley

I've been trying this with two different boards, Tiny G and Rambo. Both have similar issues. I initially set up a Matlab program that would read the robot at 38K and send to the boards at 115K. I did this because I was having trouble flashing the 38k to the boards. I was finally able to do this yesterday, but now I am having the same issues as before. 

I am now sending serial directly from the robot to the board, both at 38k. The issues I am seeing is the extruder backing up and stopping when it shouldn't. It's almost like it is reading a reverse signal but all my values are relative and greater than the previous. 

Any diagnostics you can think of that I can use to figure out what's going on?

chilipeppr, 2017-05-04:
I would recommend you pump your commands to TinyG through Serial Port JSON Server. Then you can also open ChiliPeppr just to watch the stream of commands and more importantly watch the buffer flow data. My gut tells me you're overflowing the TinyG command buffer.

rchisena, 2017-05-04:
In working with this, I have that feeling also. Is there anything that can be done if that was the case or so I just need to slow the entire process down?

johnlauer, 2017-05-04:
If you use SPJS as the intermediary then it has its own perfectly debugged bufferflow algorithm built in. Or you can go build your own.

ril3y, 2017-06-30:
You can also use this:
https://www.npmjs.com/package/g2core-api

closed 2017-06-30 by ril3y