synthetos/TinyG #203: Serial port JSON server connecttion fail

shczm, 2017-05-04:
Hi,
I download the Serial port JSON server  and run it , It said:  
staring server and webscoket on 192.168.7.105:8989
the Serial port JSON server is now running

Then I put 192.168.7.105:8989 into the chilipeppr and clicked connect, after several seconds, it said:
Failed to connect to host.
Serial port ajax error

Could anyone help me?
Thanks a lot

johnlauer, 2017-05-04:
Maybe a firewall blocking it? Are you running both on the same computer?

shczm, 2017-05-05:
Thank you for your reply , Johnlauer.
I tried again after closed the firmware, the error remains,
I am not sure if I do it in a correct way:

1. download the serial-port-json-server.exe, the dubble click on it,
2. When the conmmand line window appears, find out the IP address and put it in the chilipeppr ,then click connect.

I use win10.

Thanks

chilipeppr, 2017-05-07:
Yup. Did you figure it out? Are you running it on the same computer? Or separate computers like many folks do?

shczm, 2017-05-18:
Hi John lauer,
I think the reason was that I used a Proxy server.
So I want to use stand alone chilipeppr. However, I followed the instructions of the "chilipeppr-offline" page, it did not work properly.

[image]

chilipeppr, 2017-05-18:
You'd have to post to that chilipeppr-offline Github for help. Not sure how much they support it.

shczm, 2017-05-19:
Hi John lauer,
Finally I got the online version of chilipeppr connected to the serial port JSON server.
However it seemed that it did not work: 
I could not jog, 
If I loaded load a G code file , It could not run it

[image]

Thanks

chilipeppr, 2017-05-19:
Look at how your serial port says programming port. That's wrong

DanalEstes, 2017-05-19:
Does it work online?  If it DOES work online, and DOES NOT work offline,
please open an issue on the ChilipepperOffiline repository.

chilipeppr, 2017-05-19:
> 25112766/26228219/73895f6e-

That has nothing to do with it. Your due isnt plugged into the correct port.

shczm, 2017-05-19:
I downloaded the firware from here:
[image]
is that a right firmware?

chilipeppr, 2017-05-19:
Move your USB cable to other port

shczm, 2017-05-19:
the Serial port JSON server said:
[image]
but the conmection is fail

[image]

DanalEstes, 2017-05-19:
Sorry, John, I crossed this thread with another one from the other day.  As
you say, offline has nothing to do with this discussion.

closed 2017-06-30 by ril3y