synthetos/TinyG #205: How to Tune Motors with TinyG

routerboiz, 2017-05-15:
Hi, I am a high school student in an Engineering class. My group and I are building a CNC Router out of a TinyG, 3 NEMA 23 Motors, and a spindle speed controller. We have got our motors running on jog mode through the Universal G-Code Sender, but we cannot figure out how to send codes to the TinyG through either the Universal G-Code Sender or through Coolterm. We have tried to send $100, $101, and $102 GRBL codes to edit the steps/mm, but we have never successfully sent the codes. We have also tried other codes to tune our motors, but neither our group nor our teacher can figure out what codes to send, or where to send them. If you have any ideas, please send them our way! Thanks.

troisiemetype, 2017-05-15:
Hi,

As far as I know, I don't think that TinyG recognise these commands. You should instead use those listed on the TinyG wiki. I've been using them for three years now, without any problem, with Coolterm. I don't know Universal G-Code sender.
Setting the step per mm on motor 1, you should write $1sa=1.8 (which is 1.8 degree per step, or 200 steps per revolution)
See [TinyG command line](https://github.com/synthetos/TinyG/wiki/TinyG-Command-Line) and [TinyG configuration](https://github.com/synthetos/TinyG/wiki/TinyG-Configuration-for-Firmware-Version-0.97).

Hope it will help!

Pierre-Loup.

chilipeppr, 2017-05-15:
Or use ChiliPeppr and pull up the config widget to tweak motor settings.

[image]

[image]

closed 2017-06-30 by ril3y