synthetos/TinyG #208: TinyG G28.1 & G28.2 values are not converted when switching between G20 & G21 units.

tomauga, 2017-06-11:
I've noticed that saved workspace offsets (ie G54) are saved in the current units and translated when units are changed. So when in mm (G21), if you set G54 X = 1mm and switch to inch (G20) then $G54 is reports x offset = 0.0393 inches. Perfect.

However, if you set a G28.1 value to 10 mm and switch to G20, then $g28 reports the value as 10".

This causes problems in current sender apps. If you save a G28.1 or G28.2 position while in mm and then run a gcode file that selects inches (and leaves the app in inch mode), then going to one of those positions will not return the physical location you had hoped for.

Tested this on a TinyG with version 440.20.

-Tom

cheton, 2017-06-11:
Related issue: https://github.com/cncjs/cncjs/issues/165

ril3y, 2017-06-30:
Looking into this. @tomauga @cheton  @aldenhart  @giseburt

ril3y, 2021-04-10:
Closing due to inactivity.

closed 2021-04-10 by ril3y