synthetos/TinyG #209: Creating a artificial jog but having issues

Robdudem, 2017-06-23:
Hello Everyone,

    The problem I'm having is when I start sending a continuous stream of "G01" commands to the tinyg, (For example G01 x0 f25 -> g01x1f25 ->g01x2f25-> etc) the motors would run the first line command, pause, then move continuously. 
    What I'm trying to understand is why does it always pause after the first line command but then move continuous while commands are being pushed to the tinyg and how I can get ride of it so when I start sending gcode commands continuously the motors move continuously?  

Using a Raspberry Pi model B 
Writing the program in python 
sending to the tinyg over usb

ril3y, 2017-06-30:
How many times are you sending this?  I suspect you are either overflowing the buffer or your are not sending the commands fast enough.

Robdudem, 2017-07-01:
Twenty four times the commands gets sent from zero in one direction due to the command being incremented each time. A full range of forty eight commands from limit to limit.

ril3y, 2017-07-06:
@Robdudem try this.. instead of 1 send 5 or 10

g1x5f25
g1x10f25
...
...

I suspect it will not pause at all and continue.  TinyG is planning the first move and then nothing else comes it so it "stops" then the next consecutive moves start to fill the planner thus it does not "stop" between the moves.

This is just how TinyG works.

Robdudem, 2017-07-12:
Okay so I miss understood you ril3y before.  After figuring out you were suggesting to just throw the motion to the limit and have the tinyG move the motors to that position, while using the feedhold ! and drop moves % when I needed it to stop. Thank you.

So for anyone in the near future to create a jog on the tinyG, just send one motion command to the limit ( so in my case g00x8) and when you need it to stop send a  "b'!%\n'"(this is the command I send using pyserial ) to stop and throw out the remaining moves.

ril3y, 2021-04-10:
Closing due to inactivity.

closed 2021-04-10 by ril3y