synthetos/TinyG #211: Chaining 6 TinyG's to control 22 steppers from RaspberryPI-3

tiffyreeser, 2017-08-13:
I cannot find any reference to physically serially connect and to programmatically address individual boards and steppers. It also appears you changed from RS485 to SIO in R8. I am building a commercially available product with 22 steppers. 
Code examples from your QA or engineering lab would be great additions to your web site.

johnlauer, 2017-08-13:
For what it’s worth, I’d recommend connecting them via SPJS, not device to device as I see no way of buffering gcode from one device to the next without an intermediary like SPJS and then some code on the Raspberry Pi.

tiffyreeser, 2017-08-13:
Agree about SPJS. I cannot find any reference to a logical way or physical (jumpers) to ID the board so that I can send gcode to a particular stepper on a particular board.

DanalEstes, 2017-08-14:
Also, depending on what you are doing with all of those steppers and the
physical/mechanical requirements for them to coordinate with each other to
produce correct motion...  I foresee that the various TinyG's "planners"
and "executors" will not (cannot) stay in sync with each other.

Phrased another way, groups of three or four motors attached to a single
TinyG produce an X,Y,Z move or an X,Y,Z,A move that is predictable,
repeatable, and moves in "diagonals" (or curves, depending on the G-Code)
in 3-space or 3+A-space.  However, with multiple TinyG boards, it will not
be possible to produce a move that requires motor 1 and motor 20 to move in
such a way that they derive a consistent result in relation to each other.

At least not by feeding multiple lines of G-Code to the various TinyGs.  It
might be possible, barely possible, to feed a single line to TingG0, a
single line to TinyG1, all the way to TinyG5 (need six TGs for twenty two
motors) and WAIT for those moves to FINISH before feeding another single
line to each TG.  Even this is not going to be truly reproducible, due to
IP, USB, and/or Serial variability.   So the mechanics of the machine
cannot depend on motor groups larger than 4 having a movement relationship
to each other.

Are you willing to state, even in very general terms, how the motors
interact in the proposed machine?

Danal

tiffyreeser, 2017-08-14:
The steppers are for a display (all I can say). They are all independent and do not need to be coordinated as in a CNC. I would like to give a command to move a certain number of steps to one stepper, then without waiting for a response, give commends to other steppers so that several steppers will be executing at a time.  I would like to get or to ask for completed move responses.

tiffyreeser, 2017-08-14:
How do I ID the boards and steppers (Board 3, Stepper 2)

DanalEstes, 2017-08-14:
Hi!  I'm Danal.

To be clear, I am NOT a "core contributor" to TinyG like Alden or Riley...
nor to Chilipeppr like John... however, I have contributed to both
Chilipeppr's TinyG workspace (quite a bit) and to the TinyG firmware (a
little bit), and I build other electronics and motion control systems.

Also to be clear, to my best knowledge, neither the TinyG itself, nor the
current implementation of SPJS will "ID the boards".  As for which stepper,
that's in the G-Code.

Given what you seem to be requesting, it should be very straightforward to
plug six TinyG into a USB hub that leads to the RasberryPI, enumerate them
on the RasberryPI, and, given that SPJS is open source, modify it just
slightly to talk to all six.  Then whatever code is generating the moves
could easily "ID the board" via the SPJS (modified) API.  I recommend this
approach.

Also, a word about the G-Code and "ID ing the stepper".  In G-Code, a given
move can specify a new relative or absolute coordinate for X, Y, Z, A
and/or any combination of them.  If a multi-axis move is specified, such as
"move X one (inch or mm) and move Y twos" then the speeds will be
calculated by the motion planner so that the "control point" arrives at the
new coordinate, moving to that coordinate at the requested "feedrate" (set
in an earlier G-Code command). In my example, X would move more slowly than
Y, in such a way that the "control point" moves at the feedrate
(accelerates to the feedrate, moves at the feedrate, and accelerates to a
stop, if this is the only move in the planner).

Danal

By the way, this is probably not appropriate for a Github issue list.  I am
danal (dot) estes (at) gmail (dot) com if you wish to continue it
privately.

tiffyreeser, 2017-08-14:
True, I was planning on a USB port expander as a fallback, it is rather simple using SPJS. Thanks.
But, Synthetos does mention daisy chaining using SPI.... just have to keep digging....
Thanks. creeseratgmail

closed 2017-08-14 by tiffyreeser

DanalEstes, 2017-08-14:
Oh, and regarding "ID the boards" in SPI:  The SPI protocol specifies a "SS" Slave Select pin.  The "master" uses this pin to select which "slave" on the SPI bus pays attention to the data on the bus.  Note it is a "not" pin (there is no overbar font here), so it has to be pulled LOW to cause a given slave to pay attention. 

This is all hardware, and all standard SPI; you'll find nothing in the TinyG code.  You'd use GPIO pins on the PI for this. 

See:
https://en.wikipedia.org/wiki/Serial_Peripheral_Interface_Bus

Particularly:
https://en.wikipedia.org/wiki/Serial_Peripheral_Interface_Bus#/media/File:SPI_three_slaves.svg

chilipeppr, 2017-08-14:
I think you can just send $id via SPJS to each serial port to get back a TinyG ID. 
[id]  TinyG ID                    2X2660-FHZ

Then you could just map that via code to the appropriate board to line things up.

SPJS does give you VID/PID info from the serial ports so you can nicely iterate the serial ports and auto connect to all of them on startup of your script.

ril3y, 2017-08-16:
This is not how it works. There is no independent motion control while a
move is executing on a single board.  If you were send the x motor moving
then fired off a move for the y. The y will not start moving until the x
move has completed. However if you were using another board this would be
possible but have the same limitations on that board with single moves.

As far as chaining via spi that work was years back and I think Alden
concluded that there was drift in the timing. Perhaps not an issue for you
but it was for us.