synthetos/TinyG #212: Not New: Serial port JSON server connection fail

GaylordLivingston, 2017-08-16:
Screen Shot 2017-08-16 at 1.46.51 PM.png
Screen Shot 2017-08-16 at 1.46.09 PM.png
Screen Shot 2017-08-16 at 1.45.49 PM.png

This is still a problem. Everything was working fine. Yet a simple Pwr off and on resulted in the inability to connect. The set up is mult-step. It takes time. Even starting from scratch does not reproduce the positive results of the first go. I was so excited... Now stressed as I had a project I needed to get done. I'm running OS Sierra, Tiny G V8. I verified Incoming connections are allowed. I have swapped USB ports. Rebooted, reset, re-scanned, cleared, re-powered, rewritten, re-downloaded, deleted and started over...arghhh. An ajax error??? From where did it come> Could someone help me please.

chilipeppr, 2017-08-16:
None of the screenshots are coming through. My first guess is your OS is just not detecting the serial ports? Thus SPJS doesn't see anything?

ril3y, 2021-04-09:
closing due to lack of activity

closed 2021-04-09 by ril3y