synthetos/TinyG #213: Is TinyG still under development?

Mark-81, 2017-09-02:
Hi! I just wonder if TinyG is still actively maintained. Last commits were pushed several years ago and most issues are still opened.

aldenhart, 2017-09-03:
Yes, TinyG is still under development, but most of the work is going into g2core at this point. We intend to back-port some of the g2core features - those that can run in the more limited CPU and memory space the Xmega offers

We are also open to any pull requests if you have something you want to contribute

Mark-81, 2017-09-03:
Thanks a lot for your answer. 

I wonder if g2core has (or will have) a real "laser mode" like gbrl. TinyG is a great tool but IMHO lacks this important feature. There is a forked firmware (https://github.com/patyork/TinyG) that reduces the issue even it doesn't fix it completely.

I have some programming skills on xmega and I'd glad to contribute. But I'm just afraid that if no one has been able to fix this in three years (https://github.com/synthetos/TinyG/issues/114) hardly I would be!

robotgof, 2019-12-27:
We you still planning on back-porting anything to TinyG?  seems like the code has gone stagnant with the last revisions in September 2015 (Testing Arcs).  You mentioned back-porting in 2017.

ril3y, 2021-04-09:
TinyG is no longer in active development.  We accept pull requests!  We have moved on to continue to develop G2core where tinyG left off.  https://github.com/synthetos/g2

closed 2021-04-09 by ril3y